riscv/riscv-arch-test #283: RISCV Compliance Test for Vector Extension

HamzaShabbir517, 2022-09-19:
Hello,
To whom it may concern, I would like to know that do we have any compliance test for RISCV Vector Extension. 

Regards:
Hamza Shabbir

allenjbaum, 2022-09-19:
There are some under development, but I don't know exactly what shape they
are in.
I believe they were written for a specific configuration, and they are
attempting to modify it to be more general.

Imperas, 2022-09-19:
@HamzaShabbir517 
We have many different test suites for privilege and un-privilege subsets of RISC-V ISA. Many are available for free as open source. The latest versions of these 86 test suites are available following these links: https://github.com/riscv-ovpsim/imperas-riscv-tests/tree/v20220722/riscv-test-suite.
Note that we also include stable versions of the RISC-V tests developed by the Architectural Test Group in RISC-V International - and we provide them in the rev2.0 Bash/Make framework so they are easy to install and run.

For those parts of the RISC-V ISA that require tests that need configuring, for example, PMP, or vector instructions we license them commercially configured to the users specific core configuration.

Our vector test suite has been available for 3 years and can be configured for vector version 0.7.1 up to the current ratified 1.0.

Look for our website etc. for more information.
SimonD (Imperas)

allenjbaum, 2022-12-08:
Vector tests are under active developments AND open source tests are available, so closing this issue.

closed 2022-12-08 by allenjbaum