riscv/riscv-arch-test #288: Status of tests for bit manip extension ?

rbarzic, 2022-10-06:
Hi,

I had a look at some tests for the bit manipulation instructions and I found something strange.
For example, for the instruction andn, (defined as  X(rd) = X(rs1) & ~X(rs2)) , I can find the following test

```
inst_0:
// rs1 == rs2 == rd, rs1==x31, rs2==x31, rd==x31, 
// opcode: andn ; op1:x31; op2:x31; dest:x31; op1val:0xffffffff;  op2val:0xffffffff
TEST_RR_OP(andn, x31, x31, x31, 0xffffffff, 0xffffffff, 0xffffffff, x1, 0, x2)
```
(file riscv-test-suite/rv32i_m/B/src)/andn-01.S , line 44)

If I follow the definition of the TEST_RR_OP macro (arch_test.h)
`#define TEST_RR_OP(inst, destreg, reg1, reg2, correctval, val1, val2, swreg, offset, testreg)`

the result is expected to be 0xffffffff, with rs1 = 0xffffffff and rs2 = 0xffffffff
but 0xffffffff & ~0xffffffff  is obviously not 0xffffffff 

We found several strange "errors" like that for bit manipulation instructions so we are wondering if those tests are actually "valid" or just simple placeholders for future development

pawks, 2022-10-06:
The expected results are place holders and should be ignored. The tests themselves are correct. Correct results should be obtained by running them on SAIL.

rbarzic, 2022-10-06:
Does this apply only to the Bit Manipulation extension tests ? We have been using rv32i_m/C and rv32i_m/I for some designs for a while without the need to generate correct result with SAIL

pawks, 2022-10-06:
> Does this apply only to the Bit Manipulation extension tests ? 
No. 
**All** tests need to be used with the SAIL model. Although some of the values in the tests may be correct. They should not be treated as golden and they are not a replacement for generating values from SAIL. They are merely a feature to aid in debugging in the future.

allenjbaum, 2022-12-08:
Closed as comments now reflect are just placeholder., even if some value is passed in the macros (but is unused)
Closing this

closed 2022-12-08 by allenjbaum