riscv/riscv-arch-test #289: No c.flw or c.fsw in compressed testcases

ssecatchseagate, 2022-10-11:
The floating point directories don't have any compressed instructions and the compressed directories don't have any floating point instructions, so the compressed floating point loads and stores aren't currently covered. 

From 16.1 Overview:
" If the
C extension is implemented, the appropriate compressed floating-point load and store instructions
must be provided whenever the relevant standard floating-point extension (F and/or D) is also
implemented."

allenjbaum, 2022-10-12:
Hmm, as far as I can tell, that is may be because there is a missing
support in Sail.

allenjbaum, 2022-10-18:
No, the support is in Sail; it appears that it was overlooked because it is shared by both F and C, and it looks like each group thought it was being dealt by the other.

neelgala, 2023-01-26:
need a new set of coverpoints to get this up.

davidharrishmc, 2023-10-12:
To be precise, the following compressed floating-point loads and stores are not yet tested.

c.flw, c.flwsp, c.fsw, c.fswsp, c.fld, c.fldsp, c.fsd, and c.fsdwp

UmerShahidengr, 2023-10-12:
It has been a point of discussion for quite some time now. Even since these compressed instructions are categorised into different extensions (like Zca, Zcb, Zcf etc.). Its Sail support and ACTs are in development phase and still pending. This SoW (to provide Zc* extension support and its ACTs) was assigned to IITM at DevPartners.

jjscheel, 2024-04-01:
@ssecatch-w, Umer is correct that the work is being done by IITM.  The [DevPartners issue #6](https://github.com/riscv-admin/dev-partners/issues/6).  This work is not yet done, but has recently begun being driven again by Anusha (@anuani21).

Please read the issue for status and then reach out to Anusha and ensure that everything is being addressed or will be addressed.

UmerShahidengr, 2024-12-30:
Tests have been added in #497 , so closing this one

closed 2024-12-30 by UmerShahidengr