riscv/riscv-arch-test #292: A few tests issues

algrobman, 2022-11-04:
while porting these tests I've found following issues, would like to report: 

1) I've found that ebreak.S test runs c.ebreak instead, my gnu assembler selects smaller instruction ..
I used .option norvc to fix this.
2)  All Privilege directory tests require  rvtest_mtrap_routine define. Is it selected automatically in RISCOF?
3) when I added this define to model_test.h, fence-01.S stopped to compile , assembler complained of missing mtrap_sigptr label ..

pawks, 2022-11-04:
>     1. I've found that ebreak.S test runs c.ebreak instead, my gnu assembler selects smaller instruction ..
>        I used .option norvc to fix this.

These tests aren't supposed to be compiled with the `C` extension enabled. However, this does seem like a valid problem and the `.option` should be added to the tests. 
 
>     2. All Privilege directory tests require  rvtest_mtrap_routine define. Is it selected automatically in RISCOF?
> 
>     3. when I added this define to model_test.h, fence-01.S stopped to compile , assembler complained of missing mtrap_sigptr label ..

Yes. The tests are configured based on the configuration of the implementation. The `rvtest_mtrap_routine` includes the trap handler for these tests. The `mtrap_sigptr` label is defined in the signature section for all the tests([here](https://github.com/riscv-non-isa/riscv-arch-test/blob/8307742af42d28d44035ff115fdf91b2d1f6942f/riscv-test-suite/rv64i_m/privilege/src/ecall.S#L57)). I don't see why you are facing this error.

algrobman, 2022-11-04:
fence-01.S test is missing this code ... 
BTW, where the `#define rvtest_mtrap_routine` should be selected?

pawks, 2022-11-04:
> fence-01.S test is missing this code ... BTW, where the `#define rvtest_mtrap_routine` should be selected?

It is intended to be defined via the `-D` argument of the toolchain.

algrobman, 2022-11-04:
all this is strange and inconsistent  - c.ebreak has this define inside the test source, but a few others including ebreak don't, why?
all tests, but fence have #ifdef rvtest_mtrap_routine , why?

pawks, 2022-11-04:
> all this is strange and inconsistent - c.ebreak has this define inside the test source, but a few others including ebreak don't, why? 

This is probably an oversight while transitioning from the v2 to v3 of the framework. To get all the "privilege" tests to work, the macro was force defined at the start of the test. During transition, the macros in the other tests were removed but the c.ebreak remained. It should be removed from c.ebreak too. 

> all tests, but fence have #ifdef rvtest_mtrap_routine , why?

Other tests are mostly generated. This particular test was handwritten and possibly not updated over time.

neelgala, 2023-01-26:
a simple solution is to add .option norvc just after RVTEST_BEGIN in each test. Probably best solved in CTG wrt to the current test-suite.

allenjbaum, 2023-01-26:
The new arch_test.h does have .option norvc in the RVTEST_CODE_BEGIN macro

allenjbaum, 2023-02-09:
To put everything in one place:
- ebreak.S  should now use the correct instruction ; tests now default to .option norvc in the RVTEST_CODE_BEGIN macro.
-  Tests that can potentially trap should define  rvtest_mtrap_routine in their RVTEST_CASE string as 
  ;def rvtest_mtrap_routine. (****This is the way it looks now; this is inconsistent with the above statement
             It is intended to be defined via the -D argument of the toolchain.*the ***
- Tests that cannot trap should not have  rvtest_mtrap_routine defined; Fence and Fence.i  are examples of that.
- We should be consistent on conditionalizing the trap signature with rvtest_mtrap_routine.
--- c.ebreak and ebreak tests should be updated to  remove the #ifdef  rvtest_mtrap_routine/#endif because  rvtest_mtrap_routine is already defined and they always trap. (though it is harmless) 
 - The new trap handler should fix the issue of not compiling when rvtest_trap_handler is not defined

allenjbaum, 2023-02-23:
To clarify the current state:
- ebreak.S now used the correct instruction
- tests that trap (which are in the priv test directory) all define rvtest_mtrap_routine in their RVTEST_CASE string (including ebreak, c.ebreak and ecall) but should define it nowhere else
-  tests that don't trap do not define rvtest_mtrap_routine at all (e.g. fence and fencei)
-  tests should correctly compile when it is not defined.

This will be closed after all extraneous (but harmless)  #ifdef rvtest_mtrap_routine and matching #endif lines are removed from all tests.

allenjbaum, 2023-03-24:
> I do not see a commit in the tree for this?

Regarding these 3 specific problems listed in issue#292:

   1. I've found that ebreak.S test runs c.ebreak instead, my gnu assembler
   selects smaller instruction ..
   I used .option norvc to fix this.

I don't think we necessarily labelled the commit as fixing this, but the
latest ecall and ebreak tests have been modified to disable compressed ops.

   1. All Privilege directory tests require rvtest_mtrap_routine defined.
   Is it selected automatically in RISCOF?

Not as far as I can tell; test writers are expected to use the clause "def
rvtest_mtrap_routine" in the 2nd parameter of the rvtest_case macro for any
test that is expected to trap. Pawan can correct me if there is some
automation for that.

   1. when I added this [ rvtest_mtrap_routine] define (to model_test.h,
   fence-01.S stopped compile , assembler complained of missing mtrap_sigptr
   label ..

 In the currently merged arch_test.h file, fence.i doesn't trap, therefore
doesn't define rvtest_mtrap_routine,.
All tests that don't trap) (now) have code that defines the label
mtrap_sigptr when rvtest_mtrap_routine is not defined, so it should be fixed

allenjbaum, 2023-03-27:
Regarding #1: if riscof is used, it automatically disables the use of compressed ops in an implementation that doesn't support the C-extension. This fixes all of the problems, so I'm closing this issue.

closed 2023-03-27 by allenjbaum