riscv/riscv-arch-test #297: Suggestion: please add expected results to each RV32F test

NilsGraf, 2022-11-29:
There seems to be an inconsistency between integer and floating point arithmetic tests. For example, the first tests of the integer `ADD` and the floating point `FADD.S` are copied below:  

From [rv32i_m/I/src/add-01.S](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/I/src/add-01.S#L34-L37):
```c
inst_0:
// rs2 == rd != rs1, rs1==x4, rs2==x24, rd==x24, rs1_val > 0 and rs2_val > 0, rs2_val == 1, rs1_val == (2**(xlen-1)-1), rs1_val != rs2_val, rs1_val == 2147483647
// opcode: add ; op1:x4; op2:x24; dest:x24; op1val:0x7fffffff;  op2val:0x1
TEST_RR_OP(add, x24, x4, x24, 0x80000000, 0x7fffffff, 0x1, x3, 0, x18)
```
From [rv32i_m/F/src/fadd_b1-01.S](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/F/src/fadd_b1-01.S#L38-L44):
```c
inst_0:
// rs2 == rd != rs1, rs1==f13, rs2==f22, rd==f22,fs1 == 0 and fe1 == 0x00 and fm1 == 0x000000 and fs2 == 0 and fe2 == 0x00 and fm2 == 0x000000 and  fcsr == 0x0 and rm_val == 7   
/* opcode: fadd.s ; op1:f13; op2:f22; dest:f22; op1val:0x0; op2val:0x0; 
   valaddr_reg:x3; val_offset:0*FLEN/8; rmval:dyn; fcsr: 0;
   correctval:??; testreg:x2 
*/
TEST_FPRR_OP(fadd.s, f22, f13, f22, dyn, 0, 0, x3, 0*FLEN/8, x4, x1, x2)
```

The biggest difference between the two test styles is that the floating point arithmetic tests don't show the expected results (which is unlike the integer arithmetic test). Specifically:
*  The integer `ADD` test verifies the case `0x7fffffff + 0x1` which is expected to return `0x80000000` (see the arguments of `TEST_RR_OP`).
* The floating point `FADD.S` test verifies the case of adding `0 + 0`, where these two numbers are expressed as floating point numbers with sign `fs`, exponent `fe`, and mantissa `fm` as follows:
  *  fs1 == 0 and fe1 == 0x00 and fm1 == 0x000000  (see the first comment in the code above)
  *  fs2 == 0 and fe2 == 0x00 and fm2 == 0x000000  (see the first comment in the code above)
* However, the floating point tests don't show the expected value of each test case. Note that the comment in the code above reads `correctval:??` (note the question marks) instead of actually saying what the expected value of this test is. 

For debug purposes, it would be very helpful if each floating point test would specify its expected value, similar to the integer operations. Thanks!

NilsGraf, 2022-11-29:
For reference, the older version of the test suite also doesn't seem to show the results of individual floating-point tests. However, it uses a more compact expression of the floating point numbers as 32-bit hex values (e.g. `op1val:0x7a260524`) [Copied from here](https://github.com/riscv-non-isa/riscv-arch-test/blob/old-framework-2.x/riscv-test-suite/rv32i_m/F/src/fadd_b10-01.S#L36-L39):

```c
inst_0:
// rs1 == rs2 == rd, rs1==f0, rs2==f0, rd==f0, fs1 == 0 and fe1 == 0xf4 and fm1 == 0x260524 and fs2 == 0 and fe2 == 0x00 and fm2 == 0x000000 and rm_val == 0  
// opcode: fadd.s ; op1:f0; op2:f0; dest:f0; op1val:0x7a260524; op2val:0x7a260524; valaddr_reg:x16; val_offset:0; rmval:0x0; correctval:0; testreg:x18
TEST_FPRR_OP(fadd.s, f0, f0, f0, 0x0, 0, x16, 0, x17, x15, 0, x18)
```

allenjbaum, 2023-02-09:
Because some tests don't have a static expected result (e.g. one not dependent on which architectural options are implemented) we will not give expected results unless is easy to implement. As mentioned in issue #290, Someone interested in a result can dump the entire signature using the debug option in riscof.

closed 2023-02-09 by allenjbaum

NilsGraf, 2023-02-09:
I see. One more note on this: the [official unit tests for the F-extension](https://github.com/riscv-software-src/riscv-tests/blob/master/isa/rv64uf) have hardcoded static expected results, see for example the [unit test for fadd.S](https://github.com/riscv-software-src/riscv-tests/blob/master/isa/rv64uf/fadd.S) copied below:
```c
  #-------------------------------------------------------------
  # Arithmetic tests
  #-------------------------------------------------------------

  TEST_FP_OP2_S( 2,  fadd.s, 0,                3.5,        2.5,        1.0 );
  TEST_FP_OP2_S( 3,  fadd.s, 1,              -1234,    -1235.1,        1.1 );
  TEST_FP_OP2_S( 4,  fadd.s, 1,         3.14159265, 3.14159265, 0.00000001 );

  TEST_FP_OP2_S( 5,  fsub.s, 0,                1.5,        2.5,        1.0 );
  TEST_FP_OP2_S( 6,  fsub.s, 1,              -1234,    -1235.1,       -1.1 );
  TEST_FP_OP2_S( 7,  fsub.s, 1,         3.14159265, 3.14159265, 0.00000001 );

  TEST_FP_OP2_S( 8,  fmul.s, 0,                2.5,        2.5,        1.0 );
  TEST_FP_OP2_S( 9,  fmul.s, 1,            1358.61,    -1235.1,       -1.1 );
  TEST_FP_OP2_S(10,  fmul.s, 1,      3.14159265e-8, 3.14159265, 0.00000001 );

  # Is the canonical NaN generated for Inf - Inf?
  TEST_FP_OP2_S(11,  fsub.s, 0x10, qNaNf, Inf, Inf);
```

allenjbaum, 2023-02-10:
Those are in the riscv-tests repo, which is not the official
riscv-arch-test repo and is not compatible with the ACTs.