riscv/riscv-arch-test #299: Misaligned floating-point signature causes hang

davidharrishmc, 2022-12-03:
I recently pulled the latest repo (first time since July) and riscof now hangs when I run.  I've also updated to the latest riscof but not changed any of my other files such as the linker scripts.

The hang occurs on rv32 D tests.  The root cause seems to be misaligned doubles in the signature.

For example, I traced the problem to spike hanging on fadd.d_b1-01.S.  The hang is caused by a misaligned store address on the first test case.  The store is to 8000b614, which is not a multiple of 8. The Spike trace and disassembled code are shown below.

I added .align 3 before signature_x1_0, which fixed the issue.  But there must be something more fundamental wrong.  Any pointers?

Thank you,

David Harris

core   0: 0x8000012c (0x00e0b027) fsd     fa4, 0(ra)
core   0: exception trap_store_address_misaligned, epc 0x8000012c
core   0:           tval 0x8000b614
: 
core   0: exception trap_instruction_access_fault, epc 0x00000000
core   0:           tval 0x00000000
: reg 0 ra
0x8000b614

80000110:	50808093          	addi	ra,ra,1288 # 8000b614 <signature_x1_0>

80000114 <inst_0>:
80000114:	0001b707          	fld	fa4,0(gp)
80000118:	0081b707          	fld	fa4,8(gp)
8000011c:	00000113          	li	sp,0
80000120:	00311073          	fscsr	sp
80000124:	02e77753          	fadd.d	fa4,fa4,fa4
80000128:	00302273          	frcsr	tp
8000012c:	00e0b027          	fsd	fa4,0(ra)
80000130:	0040a423          	sw	tp,8(ra)

pawks, 2022-12-06:
This is an error due to incorrect definition of the size of the CANARY at the start of the signature region. I have a fix for it on [this](https://github.com/riscv-non-isa/riscv-arch-test/tree/canary-patch) branch and will soon raise a PR addressing this.

allenjbaum, 2022-12-07:
Ah- this is RV32D, so it's a 32bit canary, but D means it needs to be 64  -
tricky corner case. Will we need to watch out for that in any other cases?
(vector, packed come to mind, with an SEW>32, or even 64).

davidharrishmc, 2022-12-15:
Is there an ETA on the PR?

pawks, 2022-12-16:
It'll be up before 19 December.

PR #302 (2022-12-27): Canary patch
## Description
- Fix Canary definition to depend on sigalign.
- Fix SIGALIGN definition to work in all cases of FLEN and XLEN definitions.
- Fix inconsistencies in tests (missing trap regions and incomplete ISAs).
- Add Zicsr to ISA in RV32 privilege tests
- Modify signature size requirement to multiple of 4.

### Related Issues
- Fixes #190 
- Fixes #280
- Fixes #299
closed 2022-12-27 by neelgala