riscv/riscv-arch-test #303: Misaligned branches for IM extensions?

fsiegle, 2023-01-01:
Hello, sorry I'm new to Riscof and this test suite so maybe I have a basic misunderstanding. I have a RISC-V core with RV32IMZicsr extensions. All I and M tests pass fine against the reference simulator output but I'm struggling with the privileged extensions.

My core does not support misaligned branches (no C extension) and it traps in such cases as you would expect. However, I see from the signatures and the log files that the reference simulator does support the misaligned branches and therefore does not trap. Did I do a mistake in configuring the framework or is this expected behavior?

Thanks for your help!

allenjbaum, 2023-01-03:
Which reference simulator are you using?
Both Sail and Spike can be configured without the C extension, and in such
a case any branch or jump to a non-word aligned address should trap,
so it does sound like that configuration wasn't done correctly.

pawks, 2023-01-03:
Do all the misaligned branch tests fail or only some of them do? Could you also post the link for the plugin you are using for SPIKE/SAIL?

fsiegle, 2023-01-03:
Thank you for your quick answers. 

All misaligned branch tests fail currently. I didn't analyze them all in detail but they seem to have in common that my core writes into <mtrap_sigptr> whereas the simulator doesn't, so, I'm  pretty sure that it's just a configuration problem. By the way, with the misaligned load/stores I have the opposite problem, my core supports them in hardware while the simulator seems to trap.

Maybe my basic understanding is already wrong: I thought that it is sufficient to set up the isa and platform yaml files for my core (and the python plugin of course) and that RISCOF will then automatically configure the reference simulator. But maybe this is also correct and I just failed to set up these files correctly.

What I did:

- Following the Quickstart guide: https://riscof.readthedocs.io/en/stable/installation.html, i.e. installing the tools, testing the framework successfully as described (SAIL/Spike)

Then for my own core:

- Created a new folder, set up a new project for my own core (`riscof setup --dutname=coyote`) and cloned the tests.
- In coyote_isa.yaml, I changed the ISA to "RV32IMZicsr" and added "hw_data_misaligned_support: true". I also changed the misa information to match my core.
- In coyote_platform.yaml, I updated the addresses for mtime and mtimecmp registers.
- In riscof_coyote.py, I added everything required to call my RTL simulator (Questa), which dumps the signature file back to the test_dir.
- Did all the other steps as described in the manual (yaml validation, generation of test list, etc.)

My understanding is that by default I'm using the SAIL simulator as reference. The framework seems to work fine, all tests are executed correctly, the signatures are written and compared as expected. 

How does the configuration of the reference simulator exactly work? Looking at the auto-generated riscof_sail_cSim.py file (attached below), I don't see any specific command line parameters passed to it.

[riscof_sail_cSim.py.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/10336283/riscof_sail_cSim.py.txt)

pawks, 2023-01-03:
Sounds like you are right. The `C` extension is enabled by default in the SAIL binary which is executed. It has to be explicitly disabled by passing the `-C` argument in the command line. The default plugin in riscof does not seem to have that logic in place. Can you try it once by replacing that file with [this](https://gitlab.com/incoresemi/riscof-plugins/-/blob/fd-updates-sail-spike/sail_cSim/riscof_sail_cSim.py) one? I think that should fix the issue you are seeing with the misaligned branch tests. This issue might not manifest in spike because the spike binary takes the ISA as an argument in the command line. 

However, as far as the misaligned load-store tests are concerned, I am not sure if SAIL support for allowing misaligned mem operations is working. Although I do see a command line option for it. You can add the following piece of code after line 154 in [this](https://gitlab.com/incoresemi/riscof-plugins/-/blob/fd-updates-sail-spike/sail_cSim/riscof_sail_cSim.py) file:
```
            cmd += '-m'
```
Note that the above line will enable misaligned mem support by default without conditionally checking whether the DUT supports it. 

> Maybe my basic understanding is already wrong: I thought that it is sufficient to set up the isa and platform yaml files for my core (and the python plugin of course) and that RISCOF will then [...]

Your understanding is correct. The configuration steps for SAIL are missing on the default plugin inside riscof. They need to be updated to correctly translate the yaml fields to CLI flags. The configuration for SAIL is still in its early stages. Soon SAIL will be able to consume the ISA yaml files directly to set the appropriate configuration options.

fsiegle, 2023-01-03:
Great, thanks for the clarification. For some reason, your Python script didn't pick up the missing C extension from the ISA string and therefore didn't add the -C switch (I'll look into that later). But adding it manually worked fine and the test cases passed now! I think from here, I can figure it out myself. I will try now the -m switch.
Thanks for your work, the test cases are very helpful!

closed 2023-01-03 by fsiegle

fsiegle, 2023-01-03:
(update: misaligned load/store tests also pass now!)