riscv/riscv-arch-test #304: Compile warnings for SIGALIGN when FLEN is 64 in arch_test.h

ssecatchseagate, 2023-01-04:
New arch_test.h, pulled on Jan 3rd, now doubly defines SIGALIGN when FLEN is 64. This generates very verbose compile warnings. 
Please remove it from the == 64 clause, since the FLEN > XLEN check fully resolves the issue. 

#if FLEN==64
  #define FLREG fld
  #define FSREG fsd
  #define FREGWIDTH 8
  #define SIGALIGN 8
#else 
  #if FLEN==32
    #define FLREG flw
    #define FSREG fsw
    #define FREGWIDTH 4

  #endif
#endif

#if FLEN>XLEN
    #define SIGALIGN FREGWIDTH
#else
    #define SIGALIGN REGWIDTH
#endif

neelgala, 2023-01-26:
This should be fixed by #277

ssecatchseagate, 2023-01-26:
Looks like this doubling has been cleaned up in the latest repo. Thanks!