riscv/riscv-arch-test #305: More registers should be utilized for tests for increased functional coverage not specified in coverage definition

ssecatchseagate, 2023-01-09:
For example, in the div-01 testcase, there are 32 operations at the top that exercise "all the registers" by scrambling the Rd, Rs1, and Rs2 choices. there are another 557 operations with coverage for the values for the operands. However, all of there operations utilize x12 for Rd, x10 for Rs1, and x11 for Rs2. 

While this does cover the div functionality defined in the functional coverage, this testcase would do a better job of ensuring compatibility those registers were changed up a bit.

allenjbaum, 2023-01-12:
This would improve "hidden coverage" and it is recommended that CTG (and any other test generator) be updated to do this in the future.
This is should really be an enhancement request for CTG rather than the test suite, so should be filed there, but we'll keep this alive here until that happens.

UmerShahidengr, 2024-12-30:
CTG should be updated to do this in the future, but it is not needed for now. We are closing this one to clear the backlog.
@allenjbaum