riscv/riscv-arch-test #307: Please split the large floating point test suites up

ssecatchseagate, 2023-01-18:
The floating point test suites have some large coverage segments that cause testcases with 32k+ testcases within them.

These can be run as is, but in addition to having the manage the large files, which is particular problematic for debugging issues, they also cause very long simulation times for the DUT in the hardware simulations. (Since are simulations are farmed onto our cluster, the time of the simulation suite ends up being the execution time for the long pole simulation, which turns the suite from minutes into hours. With multiple smaller simulations, they can get farmed out more efficiently)

If they depended on state, these longer runs might result in more coverage. However, since there is no state saved, as users we need to deal with these massively large simulations without adding any value. 

As there is already a mechanism to break up tests into sub-tests (the last -01 number can be incremented), I suggest breaking these up to allow for more efficient simulation execution and analysis.

allenjbaum, 2023-01-18:
I think this is reasonable, but....work. Do you have an opinion for the maximum size a test should be?
There are two limits: the code size, and the data size, in addition to you issue, which is the number of test cases (you could have very few tests cases, but large code size if the test includes branches with large offsets) and the sparseness of the data size (a similar issue)

ssecatchseagate, 2023-01-19:
This is literally just a request that the simulations don't have too many instructions to execute so that they have less data output and runtime. I don't think the branches in the I code are an issue here, since the overall count of instructions is still not very large. 

As far as the floating point ones go, maybe they should be no more than twice the other longest simulation in terms of instruction count? Or even just splitting them into two or four would be an improvement (again, for the large ones, several of the floating point are very reasonable)

It might be appropriate to generate a histogram of the instruction count which I think would show a few clear outliers.

neelgala, 2023-01-26:
~10K instruction per test seems reasonable.

davidharrishmc, 2023-05-19:
The longest of the fp test suites is the fused {fmadd/fmsub/fnmadd/fnmsub}_b15 cases, which are just over a million lines and 102004 tests.  

I've recently upgraded to GCC 13.1.  On one machine, it is able to compile these tests when invoked by riscof.  On another machine, gcc usually (but not always) throws an error shown below.

When I trim to 50k tests, I still get the same error.  When I trim to 20k tests, it compiles successfully.  

This appears to be a capacity problem, and splitting the tests would solve the capacity issue as well as allowing jobs to be parallelized.

harris@vlsi:~/cvw/tests/riscof/riscof_work$ make -f Makefile.DUT-spike TARGET140
riscv64-unknown-elf-gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open my.elf; searched paths:
	. (current directory)
	/opt/riscv/riscv64-unknown-elf/bin/ (based on configured --prefix and --with-target)
make: *** [Makefile.DUT-spike:565: TARGET140] Error 255

Timmmm, 2024-05-01:
See #461 but also [this comment](https://github.com/riscv-non-isa/riscv-arch-test/pull/461#issuecomment-2088313884).

UmerShahidengr, 2024-10-31:
Closing this one, as #465 and #486 has resolved this issue. If any update is required, then anybody may reopen it.

closed 2024-10-31 by UmerShahidengr

jordancarlin, 2024-10-31:
@UmerShahidengr not sure if they are an issue, but should the corresponding Zfh tests also be split? Right now just the F and D tests were broken up. The Zfh files are smaller to begin with so it may not be an issue.

UmerShahidengr, 2024-10-31:
@jordancarlin yes, it can be done too. I will do it soon