riscv/riscv-arch-test #308: Usage of "satp" CSR for machine-mode-only cores

stnolting, 2023-01-27:
Hey there!

I am using the riscv-arch-test as part of a RISCOF [port](https://github.com/stnolting/neorv32-riscof) to check a [custom RISC-V core](https://github.com/stnolting/neorv32). Everything worked fine so far, but after I have updated to version 3.6.0 all the privilege tests are failing. I took a closer look at the new code base, but it is quite complex (but very well documented :+1:).

For testing I am using a simple `RV32IZicsr` configuration (machine-mode-only). The privilege tests fail because the code tries to access the `satp` CSR, which causes an illegal instruction exception as this CSR is not implemented in my core:

https://github.com/riscv-non-isa/riscv-arch-test/blob/af1d7e1746ffd1e0fb314f120cb1d183192c476d/riscv-test-suite/env/arch_test.h#L1403

The `CSR_XSATP` CSR is an alias for the `satp` CSR in my M-mode-only setup:

https://github.com/riscv-non-isa/riscv-arch-test/blob/af1d7e1746ffd1e0fb314f120cb1d183192c476d/riscv-test-suite/env/arch_test.h#L487

https://github.com/riscv-non-isa/riscv-arch-test/blob/af1d7e1746ffd1e0fb314f120cb1d183192c476d/riscv-test-suite/env/arch_test.h#L528

I have two questions here:

1. Did I oversee something? Are there (new?) specific configuration macros that need to be set in the platform's `env/model_test.h` file?
2. Is the platform required to implement the `satp` CSR? I checked the latest version of the RISC-V ISA spec. but I could not find any note saying that this CSR _has to be implemented_ (even if hardwired to all-zero) for machine-mode-only cores.

Thanks in advance! 😉

allenjbaum, 2023-01-27:
I think you found a bug; SATP is saved only if a trap occurs into a mode
that isn't M-mode (which would only be S, HS or VS modes),
but I didn't put the same guard around the restore operation. I'm a bit
surprised this wasn't caught when we ran the tests ourselves.
Which model were you running when this happened: your core, or a reference
model (and which one).

stnolting, 2023-01-27:
> Which model were you running when this happened: your core, or a reference model (and which one).

My port of RISCOF: https://github.com/stnolting/neorv32-riscof

I am using an older version of RISC-V Sail as golden reference: https://github.com/stnolting/neorv32-riscof/tree/main/bin I don't know the actual version right now, but I build the model from source on September 18th, 2022.

However, I discovered the problem because **my core** ([NEORV32](https://github.com/stnolting/neorv32)) caused an exception here. The Sail version I am using did not comply at all.

allenjbaum, 2023-01-27:
Ah, interesting; either there is a bug in Sail, or the Sai lmodel wasn't
configured to be M/U rather than M/S/U.
I need to look at the priv spec again; I'm not sure if the spec requires a
trap if SATP is accessed on a hart that has no S-mode, or if it's sn option.
I suspect it is required.

pawks, 2023-01-28:
@stnolting PR #309 has the patch which should have fixed the issue. SAIL does not support M/U type configuration currently. If no S mode is present and SATP is accessed, the core should throw an illegal instruction fault as per the spec.

stnolting, 2023-01-29:
@pawks I have tested your modifications from #309 and now everything works just fine again over here. 🚀

Thank you very much for fixing this so fast! 👍

allenjbaum, 2023-01-30:
There is still a puzzle. The Sail trapped in this case, which is reasonable
(but not required) behavior.
According to Andrew, Spike would also trap in this case.
We ran privileged tests with Sail as the reference, and Spike as the DUT,
and this should have trapped on both, and signatures did match.
But, your DUT trapped, but there was  a mismatch instead. So, why waa there
a mismatch, or why did our tests fail to mismatch??

commit 33b10ae3 (2023-01-28): Changelog Update. Fixes #308
closed 2023-01-30 by neelgala