riscv/riscv-arch-test #311: Signatures locations

algrobman, 2023-02-09:
Is there way to define a special data section for signatures instead of just .data and way to allocate specific address range for them?
How can target be instructed what memory addresses to be extracted to get the signature data into the signature file?

allenjbaum, 2023-02-10:
Yes, that is the intent of the RVMODEL_DATA_BEGIN/_END macros that are
supplied by the DUT.
The are badly named, in my opinion
The new trap handler encapsulates those inside standard
RVMODEL_SIG_BEGIN/END macros.
You should talk to Marc Karasek to find out how he dealt with this issue.

pawks, 2023-02-14:
> Is there way to define a special data section for signatures instead of just .data and way to allocate specific address range for them? 

You can use `.data.sig` in your model specific macro to redirect it to a region with that name. Then you can remap the specific section by allocating it appropriately in your linker file. Depending on whether the target region is before or after the code region you will have to add the following lines before line 6 or after line 13 in [this](https://gitlab.com/incoresemi/riscof-plugins/-/blob/master/spike_parallel/env/link.ld#L6) example linker file.
```
. = <target_address>
.sig : { *(.data.sig) }
```

> How can target be instructed what memory addresses to be extracted to get the signature data into the signature file?

This is target specific. Spike and SAIL have special label names to indicate the start and end of the signature regions.

allenjbaum, 2023-02-15:
The new trap handler uses the labels rvtest_sig_begin and rvtest_sig_end.
   ACTs will be using that label on all new tests.
There are new macros, RVTEST_SIG_BEGIN, and RVTEST_SIG_END,  that all new
tests will use.
They encapsulate and instantiate the model defined RVMODEL_DATA_BEGIN and
DATA_END macros.
 (which really should be named RVMODEL_SIG_BEGIN and RVMODEL_SIG_END).
An additional new macro, RVMODEL_TSIG_BEGIN is used to mark where the
regular signature ends and the trap signature begins.
(only test developers need to know about these new macros)
All existing tests have been updated to use the rvtest_sig_begin and
rvtest_sig_end labels.

More to the point: the existing test_format_spec will be split into 2
specs,
 - one for tests developers (still named TestFormatSpec) and
 -  another for test users (TestUserSpec - I'm not the most imaginative
namer)).
These will clearly document the model specific macros and labels that users
must define,
and give hints about how a user defines their model specific macros (which
PAwan gave in the previous email).
 That is WIP.

algrobman, 2023-02-16:
@MarcKarasek , to find signature addresses you can run RISCV nm utility  with a test elf file - it prints out labels names and values. 
Find begin_signature and end_signature. Memory values between these addresses need to be printed to the signature file as
words for 32 bit CPUs, or as doublewords for 64 bits CPUs ( controlled by signature granularity parameter)

MarcKarasek, 2023-02-16:
The issues is then how do you pass this to the tb running the sim... the
compile/run of the test case is all on one make target in the makefile.  So
you do not know until compile time what addresses will be and then need to
pass them to the run of the sim somehow.

And then how to translate this address into what the sim can dump to a
file...

How are you running your dut in what sim/tb?  Cadence, synopsis, etc..

algrobman, 2023-02-16:
My TB run script extracts these labels values with nm utility and provides them as plusargs to the simulator - this route should work with any simulators, supporting plusargs. As far as I know all commonly used SV simulators support plusargs, including free Verilator.

algrobman, 2023-02-16:
I extract the address inside my equivalent of your /home/mkarasek/cr4864/riscvarchtest/inspiresemi/bin/runtb.sh - the script (in my  case its ~400 lines perl script) supplies these extracted values as plusargs to verilog run command .. and TB/Verilog just dumps signature  in required format. (BTW  this script also compiles *.c or *.s tests sources outside of riscof and converts ELF to verilog $readmemh compatible format) .

Below is my "name extraction" function:

```
sub process_elf_names{
    my $test = $_[0];
    my ($tohost, $sig_start, $sig_end);
    my @elems;
    open NAMES, "riscv32-unknown-elf-nm $test.elf|" || die;
    while(<NAMES>){
#80002000 D tohost
        if(/ tohost/){
            @elems = split " ";
            $tohost =  $elems[0]; 
        }
        elsif (/begin_signature/){
            @elems = split " ";
            $sig_start = $elems[0];
        }
        elsif (/end_signature/){
            @elems = split " ";
            $sig_end = $elems[0];
        }
    }
    return " +tohost=$tohost +sig_start=$sig_start +sig_end=$sig_end ";
}

```

MarcKarasek, 2023-02-16:
So you run this script from the makefile that is produced by the
dut/env/dut.py file.

Your script runs once for each of your test case files (each .S), which are
generated by riscof..

algrobman, 2023-02-17:
you can make symbols extractions after a test elf creation and before running Verilog simulation inside the script, running this simulation.

allenjbaum, 2023-02-20:
We we add global begin_signature and end_signature labels, aliased to rvtest_sig_begin and rvtest_sig_end to the trap handler, does that solve your problem, or can you use those rvtest_ labels which will be defined in the new test spec?

algrobman, 2023-02-20:
> We we add global begin_signature and end_signature labels, aliased to rvtest_sig_begin and rvtest_sig_end to the trap handler, does that solve your problem, or can you use those rvtest_ labels which will be defined in the new test spec?

@allenjbaum , whom are you asking, me or Marc? If me, I use these begin/end_signature  labels already - my problem was that I didn't know what symbols to use ...

allenjbaum, 2023-02-20:
Ah, then you should use rvtest_Sig_begin and rvtest_sig_end, because that
is what the arch_test.h makes global.
We could also alias begin_signature and end_signature labels to the above
in future version if that is necessary.

algrobman, 2023-02-20:
I did use rvtest_sig_begin/end,  but got signatures diffs with Sail signature having extra zeros vs my signatures, filing issue on sail simulator ... It is something to do with these labels alignment requirements for the sail simulator. (16 bytes, I think)

allenjbaum, 2023-02-21:
Ah, that explains it; we didn't coordinate the release. I hadn't realized
that this was built into Sail. Thanks.

allenjbaum, 2023-02-24:
This should be fixed now in https://gitlab.com/incoresemi/riscof-plugins/-/merge_requests/30,

pawks, 2023-02-27:
Closing this issue as it has been addressed in the appropriate plugins. To use the latest plugin, clone/get the plugin from [here](https://gitlab.com/incoresemi/riscof-plugins/-/tree/master/sail_cSim) or update(if already cloned).

closed 2023-02-27 by pawks