riscv/riscv-arch-test #313: test compilation warnings

algrobman, 2023-02-11:
I'm getting following warnings:

```
riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cadd-01.S: Assembler messages:
riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cadd-01.S:37: Warning: shift count out of range (-1 is not between 0 and 63)
riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cadd-01.S:47: Warning: shift count out of range (-1 is not between 0 and 63)
riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cadd-01.S:62: Warning: shift count out of range (-1 is not between 0 an
```
...
is this expected?

pawks, 2023-02-13:
This looks incorrect. It is not an expected warning message.

algrobman, 2023-02-13:
And what should I do about this?

pawks, 2023-02-13:
This is mostly a problem in the LI macros. Its not due to your execution environment. I don't see any problem in the loaded values. You can ignore these warnings for now. They will be fixed soon.

algrobman, 2023-02-13:
the only problem is that if riscof fails - my terminal is filled with these warning messages, I don't see real reason for the failure ..

allenjbaum, 2023-02-14:
Which version of the arch_test.h macro are you using? I thought that issue
was fixed.

allenjbaum, 2023-02-15:
> I see this issue on the main branch of arch-test.

OK, then LI needs to be fixed; can you send an example of where that
occurred and any values that are possible to extract to cause this?
This is the first I've heard of it, so it must have something to do with
the linker relocations, right?

algrobman, 2023-02-15:
I didn't change a character in the test sources and env files .

These warnings are not seen if all tests pass, but fill shell window if any of the test fail for whatever reason, including signature comparison

allenjbaum, 2023-02-16:
Hmm, cause or effect? Can you send the failure report, and possibly the
linker scripts?

algrobman, 2023-02-16:
I did not change anything from the tests repo, including linker scripts.

allenjbaum, 2023-02-16:
If that's the case, you must be comparing Spike with Sail, and there were
no failures when we did it.
If instead you were comparing Sail or Spike vs your DUT, then you are
responsible for the DUT linker script, aren't you?

algrobman, 2023-02-16:
Again, I ran  riscof as is without changing files from the tests repo.

allenjbaum, 2023-02-16:
But you also said that you didn't see the message when all tests pass, only
when they fail,
which implies that there were test failures - but when we ran the tests
from the test suite,
we got no failures (and if we did, we would not have merged any new
changes).
So which tests failed, and what were the signature mismatches?

algrobman, 2023-02-16:
I had failure due two issues 1) my core supported misaligned load/stores, but sail was not configured with that.
2) when DUT simulations timed out/killed before I  increased the run timeout as I was advised.
this 2nd case was pretty confusing since this happened after I added B-ext tests, but the failing was a I-ext test and I couldn't see the failure reason as whole the shell window was overloaded with these warnings . 
I think you can model this  issue if you set execution timeout to let say 1s

allenjbaum, 2023-02-16:
Ah, thanks for that extra information. This starts to make sense, though I
can't say why those warnings are generated.

allenjbaum, 2023-02-20:
To reiterate: you say you are not changing any scripts, but you needed to configure Sail to support misaligned (which will be fixed at some point), and you changed the timeout value (are you running on a particularly slow server?) and you had to supply your own custom linker script. 
IT is completely unclear for the above description if those warning and failures go away when misaligned is supported and the timeout is increased or not - you are not giving us enough information.
IF you are still seeing these issues, please send us a copy of the shell window so we can see what is going on. 
If not - it would still be useful to figure out why a timeout - or an actual test failure - would cause the specific error message that you're seeing.

algrobman, 2023-02-20:
@allenjbaum , Are you reading the thread? @pawks sees the same issue. (i.e. warnings caused by LI instructions).
Assembler warnings don't kill the tests and seems the tests are simulated correctly. But 1) the tests shouldn't generate warnings if written correctly. 2) if something is going wrong, these warnings from all these tests just fill shell history, making hard to figure out the failure reason.

Regarding my updates - I'm changing only sail simulator and DUT python plugins, just adding -m option for sail run command and 
updating execution command for my RTL simulator . Running verilog simulation more than 5min for more than 100 tests seems to me quite reasonable. Again I do not change any file from the RISCV tests repository, so you should be able reproduce these warnings and fix your tests

algrobman, 2023-02-20:
if you run one test target  you can see all these warnings.

algrobman, 2023-02-20:
these warnings come from your super complex macro: (arch_test,h)
```
#define LI(reg, imm)                                                            ;\
  .option push                                                                  ;\
  .option norvc                                                                 ;\
  .set even,    (1-BIT(imm, 0)) /* imm has at least 1 trailing zero     */      ;\
  .set immx,    (imm & MASK)    /* trim to XLEN (noeffect on RV64)      */      ;\
  .set imm12,   (SEXT_IMM(immx))                                                ;\
  .set absimm,  ((immx^(-BIT(immx,XLEN-1)))&MASK) /* cvt to posnum to simplify code */  ;\
  .set imme,    ((immx^(-BIT(immx,0     )))&MASK) /* cvt to even, cvt back at end */    ;\
  .set cry,     (BIT(immx, IMMSGN))                                             ;\
  .set cryh,    (BIT(immx, IMMSGN+32))                                          ;\
  .set pos,      0                                                              ;\
  .set edge1,   -1              /* 1st "1" bit pos scanning r to l      */      ;\
  .set edge2,   -1              /* 1st "0" bit pos scanning r to l      */      ;\
/**** find 1st 0->1, then 1->0 transition fm LSB->MSB given even operand ****/  ;\
  .rept XLEN                                                                    ;\
    .if     (edge1<0)           /* looking for first edge?              */      ;\
      .if (BIT(imme,pos)==1)    /* look for falling edge[pos]           */      ;\
        .set  edge1,pos         /* fnd falling edge, don’t chk for more */      ;\
      .endif                                                                    ;\
    .elseif (edge2<0)           /* looking for second edge?             */      ;\
      .if (BIT(imme,pos)==0)    /* yes, found rising edge[pos]?         */      ;\
         .set  edge2, pos       /* fnd rising  edge, don’t chk for more */      ;\
      .endif                                                                    ;\
    .endif                                                                      ;\
    .set    pos,  pos+1         /* keep looking (even if already found) */      ;\
  .endr                                                                         ;\
  .set immxsh, (immx>>edge1)    /* *sh variables only used if positive  */      ;\
  .set imm12sh,(SEXT_IMM(immxsh))/* look @1st 12b of shifted imm val    */      ;\
  .set crysh,     (BIT(immxsh, IMMSGN))                                          ;\
  .set absimmsh, immxsh         /* pos, no inversion needed, just shift */      ;\
/*******does it fit into std li or lui+li sequence****************************/ ;\
  .if     ((absimm>>IMMSGN)==0) /* fits 12b signed imm (properly sgnext)? */    ;\
       li   reg, imm12         /* yes, <= 12bit, will be simple li       */    ;\
  .elseif (XLEN < 64 || (absimm+ (cry << IMMSZ) >> WDSGN)==0) /* fits 32b sgn imm?(w/ sgnext)?    */    ;\
        lui  reg, (((immx>>IMMSZ)+cry) & LIMMMSK) /* <= 32b, use lui/addi */    ;\
    .if   ((imm&IMMMSK)!=0)     /* but skip this if lower bits are zero   */    ;\
        addi reg, reg, imm12                                                    ;\
    .endif                                                                      ;\
 /*********** look for various masks 0s->1s->0s, etc **********************/    ;\
  .elseif ( even && (edge2==-1))        /* only rising  edge, so 111000   */    ;\
         li      reg, -1                                                         ;\
        slli    reg, reg, edge1         /* make 111s --> 000s mask        */    ;\
  .elseif (!even && (edge2==-1))        /* only falling edge, so 000111   */    ;\
        li      reg, -1                                                         ;\
        srli    reg, reg, XLEN-edge1    /* make 000s --> 111s mask        */    ;\
  .elseif (imme == (1<<edge1))          /* check for single bit case      */    ;\
        li      reg, 1                                                          ;\
        slli    reg, reg, edge1         /* make 0001000 sgl bit mask      */    ;\
    .if   (!even)                                                               ;\
        xori    reg, reg, -1            /* orig odd, cvt to 1110111 mask  */    ;\
    .endif                                                                      ;\
  .elseif (imme == ((1<<edge2) - (1<<edge1))) /* chk for multibit case    */    ;\
       li      reg, -1                                                         ;\
        srli    reg, reg, XLEN-(edge2-edge1) /* make multibit 1s mask     */    ;\
        slli    reg, reg, edge1         /* and put it into position       */    ;\
    .if   (!even)                                                               ;\
        xori    reg, reg, -1            /* orig odd, cvt to 1110111 mask  */    ;\
    .endif                                                                      ;\
  /************** look for 12b or 32b imms with trailing zeroes ***********/    ;\
  .elseif ((immx==imme)&&((absimmsh>>IMMSGN)==0))/* fits 12b after shift? */    ;\
        li      reg, imm12sh            /* <= 12bit, will be simple li    */    ;\
        slli    reg, reg, edge1         /* add trailing zeros             */    ;\
  .elseif ((immx==imme)&&(((absimmsh>>WDSGN)+crysh)==0)) /* fits 32 <<shft? */  ;\
        lui     reg, ((immxsh>>IMMSZ)+crysh)&LIMMMSK /* <=32b, use lui/addi*/    ;\
[... cut ...]
        addi    reg, reg, (immx)     & (IMMMSK>>2)      /* lst 10b (09:00) */   ;\
    .endif                                                                      ;\
    .if (XLEN==32)                                                              ;\
        .warning "Should never get here for RV32"                               ;\
    .endif                                                                      ;\
 .endif                                                                         ;\
```
BTW, just curious, why isn't  just assembly 'li' pseudo instruction sufficient?

allenjbaum, 2023-02-21:
I was quite certain that it came from LI; what I would like to know is
which immediate is causing that to happen;
I don't recall seeing it when we tested it, and looking through the code,
it shouldn't be possible (with the latest version of the macro)
It shifts by edge1 (which is the result of a count_trailing_zeros loop on
an even number), so must be in the range 1..XLEN-1
 If it all zeroes, it is -1, but the all-zeros case never shifts anything
(because it fits into a 12bit immediate)
IF there is an escape in this logic, I'd like to find it.

But, that's a good question - why our own version.
The answer is that we need these to have the same length, regardless of
toolchain,
and we couldn't guarantee that all tochains did that, or even if they did,
used exactly the same instructions.
So we guaranteed it by making our own definition.

algrobman, 2023-02-21:
> The answer is that we need these to have the same length, regardless of toolchain,

Could you please elaborate, what do you mean by "the same length" (or length of what) and why did you need this? - the goal of LI is just to set a register a value. Thus for 64 bits CPU li translates to different number of instructions than for 32 bit CPUs.  
Also the compiler can exploit constant value to use less instructions. Looking in your macro, seems it  generates different number of instructions as well, depending on the constant,  and uses different instructions too.

We always use LI/LA pseudos' in our tests with no need in special macro.

> I was quite certain that it came from LI; what I would like to know is which immediate is causing that to happen;

see initial description. the warnings  point to exact test code lines .

allenjbaum, 2023-02-21:
The same length means the same number of ops/bytes in the expansion of the
LI.
If LI generates differing lengths, (between the reference and the DUT),
 then trap signatures will have different xEPC, and xTVAL values.
Because the linker scripts can be different as well, we compensate by
adjusting EPC and TVAL (when it holds an address)
 to be an offset from the rvtest_data_begin and rvtests_code_begin labels.
But that doesn't compensate for random
different offsets caused by different sized li pseudoops. , .align
directives can also magnify the effect of that (or remove it...).

We have a fixed sized version that does a .align before, and a .align
after, but it really blows up the length of the tests,
which is why I wanted something much more compact. This isn't perfect, but
captures most of the use cases for the trap handler.
Tests are stuck with using that form of LA, though, especially as a bare
"la" won't work with different linker scripts.
The trap handler has been updated to remove almost all cases of LA from it
(that will be in the next version),
at the cost of requiring tests that create/update/modify page tables have
to update trp handler data structures to match.

allenjbaum, 2023-02-21:
What is strange is we have test code using those constants, and it compiled
correctly.
For RV32, it should always generate either  should always generate the
sequence
   addi, reg, x0, imm
for small numbers in the range -2048..2047 inclusive, or

   addi, reg, x0, (imm+(cry<<12)>>12; addi reg, reg, SEXT_IMM(imm)
otherwise, with no shifting - all that other cod in that complicated macro
is essentially for RV64 (or 128, I suppose)
I don't suppose you could produce a disassembly listing of this test...?

allenjbaum, 2023-02-21:
We think that this may be a quirk in the way assembler macros are processed; e.g. it generates all possible code paths, then chooses the right one depending on the conditions it has evaluated, but that process generates warnings that are not suppressed in code that is generated but not selected. We could fix this for RV32, but unclear if we can for RV64 (which is where the bulk of the code is used)

algrobman, 2023-02-21:
What I care about is RV32 for now. You could have two versions of the macro - one for 32 and another for 64 bits platforms

allenjbaum, 2023-02-22:
I'm working on that theory.
To be clear: you can pass all the tests when misaligned is properly
configured in sail, and timeouts are increased - but still get those
warnings, and you want the warnings removed.

algrobman, 2023-02-22:
>  but still get those warnings, and you want the warnings removed.

Exactly  - warning removal/fix was the original intent of this issue! . Other stuff is just ways to see the warnings impact on user. 
One more way to see these warnings  probably is to corrupt one of the tests to make it fail compilation ..

PR #318 (2023-04-19): Updated trap handler
This PR is co-authored by Allen Baum @allenjbaum. In this update, we have made the following changes:
- Split LI() macro for RV32 and RV64 to eliminate warning messages which should resolve #313 .
- Reduce the use of LA() macro by declaring fixed size offsets in save area.
- Updated RVTEST_GOTO_MMODE macro
- Updated \__MODE__\()trap_sig_sv, replaced an LA(rvtest_trap_sig) by calculating initial_Xtrap_sigptr + (Mtrap_sigptr-initial_Mtrap-sigptr) 
- Similar changes are made in chk_\__MODE__\()trapsig_overrun macro.
- Updated RVTEST_TRAP_SAVEAREA.
commit 1da0f46a (2023-04-19): Update LI macros, Added infra for VM traps, fixed #313, fixed #332
closed 2023-04-19 by neelgala