riscv/riscv-arch-test #314: How to run Bitmanip and Atomic  tests

algrobman, 2023-02-12:
How do I include A-ext, ZBA, ZBB, ZBC and ZBS tests to be run by riscof and be part of report?

Are there compliance tests for RV32E, ZMMUL ? and how to run them?

pawks, 2023-02-13:
There are no explicit tests for the `ZMMUL` extension. There are tests for the `M` extension. You can delete all the div tests in the directory(or edit the `database.yaml`) and then include the `M` extension in the ISA field to run those tests using riscof. 

For the ZB* extensions, you can run the tests by including the respective extensions in the ISA field of the yaml canonically.

pawks, 2023-02-22:
> There are no explicit tests for the `ZMMUL` extension. There are tests for the `M` extension. You can delete all the div tests in the directory(or edit the `database.yaml`) and then include the `M` [...]

Ignore this comment. The tests have already been fixed to account for the `Zmmul` extension. You would have to add the extension in the isa string of your yaml and riscof should work out of the box. I am not sure if the toolchains support the `Zmmul` as a part of the `march` argument so YMMV.

allenjbaum, 2023-02-23:
A-extension: no tests yet, I've started the process for creating them
Bitmap extension(s):  they exist and can be run by adding the extension names to the ISA string
Zmmul: the tests can be run from the M-extension directory by adding the Zmmul to the ISA string
RV32E: There are currently unratified RV32EMC tests in the repo (but not RV64 - though if you're worried about the size of the register file, then you probably you wouldn't want an RV64E - or even an RV32EFD). 
   Once RV32E is ratified, we will change the name and start generating tests for other extensions

 I think we given all answers and we can close this issue. Reopen if something isn't working as expecxted.

closed 2023-02-23 by allenjbaum