riscv/riscv-arch-test #315: Are Atomic instructions supported?

algrobman, 2023-02-13:
Are Atomics (A-Ext) supported? and how to configure riscof to run them?

pawks, 2023-02-13:
There are no tests for the `A` extension currently. Riscof follows the same method of configuration for all the extensions i.e using the isa yaml.

allenjbaum, 2023-02-20:
Unknown. I've created an SOW for Dev partners, tiemframe is TBD. Note that it will not be possible to test that they are multi-hart atomic, only they function correctly within a signel hart (even if interrupts occur immediately before or after.

algrobman, 2023-02-20:
First of all we need tests that check basic functionality of these instructions - like amoadd adds a register value to a memory location - these types of tests shouldn't be hard to generate. (A CPU implementer can mess up even this due the spec misunderstanding). Multiprocessor aspects of these instructions is much more difficult to verify and I'm not sure can be done in this "compliance" tests framework. (Similar as "invisible" aspects of fence/fence.i instruction or interrupts /exceptions handling compliance. )

allenjbaum, 2023-02-20:
yes - exactly.
The SOW I wrote up does exactly that (and includes some interesting values
as well)
There are lots of tests that need writing, and it's all volunteers that do
it it (some volunteered by the companies emplying them, or by university
grants) so it takes a while.

algrobman, 2023-02-20:
why can't your test generator  do this? (one, which generated all other RISCV tests)I'm running A-Ext tests from https://github.com/riscv-software-src/riscv-tests, which look very similar to the tests from your repo. ..

pawks, 2023-02-22:
> why can't your test generator  do this? (one, which generated all other RISCV tests)I'm running A-Ext tests from https://github.com/riscv-software-src/riscv-tests, which look very similar to the tests from your repo.

The test generator needs augmentation to handle the A extension tests. It does not support it in its current state.

allenjbaum, 2023-10-25:
I think the test have been merged into the repo, so I'm closing this. Reopen if you see something missing

closed 2023-10-25 by allenjbaum

algrobman, 2023-10-29:
Are there LR/SC tests? I've found AMO*  instructions tests only in A- directory ..

allenjbaum, 2023-10-30:
This is an issue that has been opened. Supposedly coming shortly.