riscv/riscv-arch-test #317: Missing illegal instruction testcase

ssecatchseagate, 2023-02-21:
There is no testcase that ensures that an illegal instruction traps. 
Unfortunately, mtval is implementation independent, but mepc should at least indicate the bad instruction location. 
This testcase should exist for both compressed and non-compressed formats.

pawks, 2023-02-22:
The only test cases possible for this particular scenario is to have a 2 byte value filled with 0's(for both 32 and 64) and a [ILEN-1:0] filled with all 1's as these are the only 2 encodings which are guaranteed to be illegal instructions as per the ISA. So 3 test cases:

- 2 bytes of all 0's (for both scenarios where C is present and Absent)
- 2 bytes of all 1's(for Implementations where C is present)
- 4 bytes of all 1's(for implementations where C is absent and cases where C can be disabled)

ssecatchseagate, 2023-02-22:
This would certainly ensure that the illegal instruction is detected and traps properly. 
I concur that every single illegal instruction for a given implementation is out of scope of compatibility testing.

allenjbaum, 2023-02-22:
Good catch.  The canonical illegal ops ( op[31:0]==all1 and op[15:0]==0) should be easy; a corner case is what is reported in xTVAL. The spec says that the minimal length encoding is detected, which I interpret to mean that C-extension is present, xTVAL reports 32bits of zero, but if it isn't the xTVAL should have the full 32bits, even if the upper bits are non-zero (which are ignored).

ssecatchseagate, 2023-02-22:
What is that from? My reading of the privileged spec doesn't require a write to mvtal and it is allowed to remain all zeros- which happens to work for the all zeros case, but not for the all ones case. (bolded the text that supports that it allowed to be all zeros, even the if the illegal instruction is something else) (Document Version 20211203)

The SAIL model also appears to behave this way. (From debugging the trap code compiled with the compressed extension- for my run it caused an illegal instruction execution, and SAIL left mtval as all zeroes). Once configuration is added, this also will need to be a setting for the YAML. 

The mtval register is an MXLEN-bit read-write register formatted as shown in Figure 3.23. When a
trap is taken into M-mode, mtval is either set to zero **or** written with exception-specific information
to assist software in handling the trap. Otherwise, mtval is never written by the implementation,
though it may be explicitly written by software. The hardware platform will specify which exceptions must set mtval informatively and which may unconditionally set it to zero. **If the hardware
platform specifies that no exceptions set mtval to a nonzero value, then mtval is read-only zero.**
...
The mtval register **can optionally** also be used to return the faulting instruction bits on an illegal
instruction exception (mepc points to the faulting instruction in memory).

allenjbaum, 2023-02-23:
I'm not sure exactly what you are disagreeing with. I didn't specifically say that mtval was loaded, but that was what I was assuming for this example. Tests don't care, they get loaded or not, and it only matters whether it agrees with Sail or not.
But when mtval is loaded on an illegal inst trap. (and JJJJ != op[15:0])  
 illegal op               tval with Cext   tval w/o Cext 
0x0000_0000       0x0000_0000   0x0000_0000    <-low 16 bit==0
0xJJJJ_0000        0x0000_0000   0xJJJJ_0000     <-low 16 bit==0, ilen=16 or 32
0xFFFF_FFFF        0x0000_FFFF    0xFFFF_FFFF     <- low ILEN bits==FFFF, for both ILEN==16 or 32
0xJJJJ_FFFF        0x0000_FFFF    don't test           <- may not trap:  JJJJ!=0xFFFF for ILEN==32
   So, 3 cases we can tests regardless of C extension, and 1 only if C-extension is implemented (and enabled)

When mtval is not loaded on an illegal inst, it is always loaded with zero when an illegal inst trap occurs
Test should always (attempt to) initialize mtval to some non-zero value to ensure that it isn't keeping the old value

ssecatchseagate, 2023-02-23:
I didn't think we can check it's the "correct value", but it seems like we should be able to check it's "not the old wrong value" with code. It seems like it would require hand written test code rather than a signature comparison, of course. I was initially expecting a signature comparison of mtval written to the signature region, thus my confusion.

allenjbaum, 2023-02-24:
Not sure what you mean about the "correct" value. The "correct" value is whatever the Sail model returns,
and that is what I've listed above (if I haven't made a mistake interpreting the spec....)
For this particular case, EPC does point to the offending instruction, and we could load that (if it is readable) and store it into the signature, but that's more work than I think we want to do (and not always possible if it isn't readable in the executing mode, so we'd have to do a software page table walk to get the opcode that failed
So why do you think this needs to be a hand-writing test instead of a signature comparison?

ssecatchseagate, 2023-02-24:
Because mtval can be all zeroes, or it can contain the address of the fault. There is no "correct" value. 

If mtval is written with a nonzero value when an instruction access-fault or page-fault exception
occurs on a system with variable-length instructions, then mtval will contain the virtual address of
the portion of the instruction that caused the fault, while mepc will point to the beginning of the
instruction.

allenjbaum, 2023-02-24:
My definition of correct is different than yours.
The correct value is zero if the architectural option to not load the
address/instruction is implemented.
It is not the case  - for a specific implementation - that sometimes it
will be zero and sometimes the address or instruction (for each exception
cause individually).

ssecatchseagate, 2023-02-24:
Sorry, note that I initially pasted the incorrect fault in my comment, which you replied to above. I fixed it on the repo. 

If there is a variable answer that is allowable, and there is currently no configuration value for that value in SAIL, then the testcase can't compare a signature..  although in this case, it would work exactly as expected for the all-zeroes, case. For the all ones, code would have to be written to branch off "This is not the value we wrote initially into mtval" to write something into a signature for us to compare, rather than to look at the signature and do a direct comparison. 

If we're just comparing epc, that can be a direct comparison, as you've already indicated. For tval, some code additional code would be need to be written for the all zeroes case to get something that can be compared, or this comparison would need to be deferred until the SAIL model can be configured to match the DUT. The "additional code" is what I was referring to as "hand written".

allenjbaum, 2023-02-24:
I'm missing something here - we are talking past each other somehow.
Misaligned can be variable, unpredictable, and nondeterministic.
Writing tval is not dynamically variable. It i either always written with
address/opcode, or always written with zero
(on an mcause-by-mcause basis.
Sail is configured to do that, and it either matches the implementation or
doesn't.

ssecatchseagate, 2023-02-24:
I think we're in agreement. I was referring to the "it doesn't" case.

allenjbaum, 2023-02-24:
I don't remember where in the thread that "it doesn't" came up (no- don't
send it to me!),
but I thought we were actually agreeing, and I'm glad that we are agreeing
that we agree.

UmerShahidengr, 2024-12-30:
We need a test for this case to close this issue.

> I'm not sure exactly what you are disagreeing with. I didn't specifically say that mtval was loaded, but that was what I was assuming for this example. Tests don't care, they get loaded or not, and [...]