riscv/riscv-arch-test #320: Linked coverage report is missing

DarrenGalpin, 2023-03-08:
The coverage report linked in to https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/README.md is missing and gives a file not found error.

pawks, 2023-03-09:
The coverage reports have been moved to a different location([link](https://drive.google.com/drive/u/3/folders/1KBRy6OgxnOPTDgyfJDj0gcMi5VdMLtVo)). It is organized in the same structure as the test suite.

allenjbaum, 2023-03-10:
Yes, I found it, and added a READ ME to describe what this google drive
folder contains.
I suppose I should have created a google doc instead of a text file for
that; it looks pretty ugly.
I guess my real question is: do test developers know they're supposed to
place those files in the test directory? Or that they should be Zipped?
OR does that happen automatically when the new tests are merged (by the C/I
script)?
And, if that's the case: why do we need separate PR and Released
directories?

UmerShahidengr, 2024-12-30:
Issue was resolved, closing this issue, in case of any problem, you can reopen it anytime

closed 2024-12-30 by UmerShahidengr