riscv/riscv-arch-test #322: Correct values in rv64i_m/K/src/<Instruction Name>-01.S are all 0s.

mlawson-tt, 2023-03-27:
These tests use helper macros to check the results of the instruction that each include the correct value after the instruction occurs, but for all of the crypto (K) tests, the correct values are all 0s.

I assume the values are 0s just because they haven't been populated yet.  Are there any near-term plans to populate the real correct values for these tests?

I did notice that these tests were generated using a different test generator (https://github.com/riscv-software-src/riscv-ctg).  If it makes sense to also file an issue with that repo as well, I can do so (though it seems like it also makes sense to keep here as well since the tests will likely need regenerated if the change occurs in the generator).

mlawson-tt, 2023-03-27:
OK, thanks.  I'm curious though how it's intended to score the results of the tests in that case.

BTW feel free to close this issue if it's not something that is a priority.

allenjbaum, 2023-03-27:
Tests are "scored" by comparing the results with the results of the same
code running on a reference model (preferably Sail).
There was a plan once to back annotate the tests with those results, but
  a) resource constraints,
  b) many tests will have different results depending on
implementation-defined configuration options, so there is no single correct
value.
  ( the reference model, configured with those same configuration options,
will  give the "correct "value for that implementation, though)

mlawson-tt, 2023-03-29:
OK, thanks, after reading your response and a couple of days of doing some research, I think scoring makes sense now.  Seems like the best (and probably intended) method of scoring these tests is to create a plugin for [riscof](https://github.com/riscv-software-src/riscof) and use Sail and/or Spike as the reference model.

Still seems like it might be worth adding the correct values in the macros eventually (in those situations where there is only 1 correct answer) just for quick sanity checks, but that's really where the utility ends, so it's not really a big deal to address anytime soon.

cmuellner, 2023-04-04:
I want to chime in on this topic with a proposal (as I also stumbled over it):
Why not state, "we don't do known-answer-tests, but we only compare signatures against the reference implementation (SAIL)" in the README.md. I think this fundamental property should be documented prominently.

And just because it does not work in all cases does not mean that KATs should not be used.
For a typical pure computational instruction (most unpriv instructions) KATs are sufficient.
Just like there are different test methodologies for devices or applications, there can also be different test methodologies for instructions.

UmerShahidengr, 2024-12-30:
It will be updated in TestFormatSpec once it will be updated. Also CTG will also be updated to give proper comments against the test cases if it will have the correct value or not. Issue #586 will handle this one. Closing this one

closed 2024-12-30 by UmerShahidengr