riscv/riscv-arch-test #326: Unclear label `rvtest_data`

cmuellner, 2023-04-03:
Many or all tests use the label `rvtest_data`, which is not documented in the `TestFormatSpec.adoc`.

Note, that the undocumented macros `TEST_LOAD` and `TEST_LOAD_F` use this label.

allenjbaum, 2023-04-03:
That's an artifact left over from older tests.
New tests will have the test data area begin with the lavel
rvtest_data_begin:, which is generated inside the RVTEST_DATA_BEGIN macro.
Old tests are still compatible with newer versions of the RVTEST_DATA_BEGIN
macro, and we haven't gone back  and removed the labels (which would
involve updating the tests.
So, they are now redundant, harmless labels.

pawks, 2023-04-10:
The TestFormatSpec is a document which talks about the guidelines for writing the tests. `rvtest_data` is not a required label and is named as such just for verbosity. It is subjective to the test writer and any label can be used to point to the memory location.

pawks, 2023-04-10:
I do agree on the missing documentation. The macros in `test_macros.h` can definitely use some additional comments to explain them.

allenjbaum, 2023-04-10:
And... as mentioned elsewhere, I am currently rewriting the test format
spec, splitting into two pieces: what a test developer needs to know, and
what a test user needs to provide to run the tests.

UmerShahidengr, 2024-12-30:
Another thread to see while updating TestFormatSpec doc (#586 )

closed 2024-12-30 by UmerShahidengr