riscv/riscv-arch-test #328: All floating-point instruction tests don't have covergroups

cmuellner, 2023-04-04:
The covergroups in the `coverage` directory are only available for instructions that operate on integer registers.

So far, there are two additional classes of registers in RISC-V:
* floating-point registers (with different FLEN for F, D, Zfh, Q)
* vector registers (with implementation-specific VLEN)

Are there plans to add support for them?

pawks, 2023-04-10:
The covergroups for the F & D extensions do exist([link](https://github.com/riscv-software-src/riscv-ctg/tree/master/sample_cgfs/sample_cgfs_fext)). However, no Vector support is available currently. Note that the coverage files in this repo are outdated and will soon be purged from the repository.

UmerShahidengr, 2024-12-30:
All covergroups have been added in riscv-arch-tests and CTG repo has been archived. You will be able to find everything in coverage folder.

closed 2024-12-30 by UmerShahidengr