riscv/riscv-arch-test #332: On RV64 the ecall and ebreak tests throw a load unaligned exception ...

InspireSemi, 2023-04-09:
In the spcl_Mhandler code it tries to do a ld (based on rv64) from the clrint_Mtbl. 

In rv64 mode this ends up as a ld t3, 0(t3), which tries to laod from a 32 bit address a 64 bit word and throws a misaligned access.

This is the disassembled code:
0000110000000800 <spcl_Mhandler>:
    110000000800:	00000e17          	auipc	t3,0x0
    110000000804:	01ce0e13          	addi	t3,t3,28 # 11000000081c <clrint_Mtbl>
    110000000808:	007e0e33          	add	t3,t3,t2
    11000000080c:	003f1393          	slli	t2,t5,0x3
    110000000810:	007e0e33          	add	t3,t3,t2
    110000000814:	000e3e03          	ld	t3,0(t3)

0000110000000818 <spcl_Mdispatch>:
    110000000818:	000e0067          	jr	t3

000011000000081c <clrint_Mtbl>:
    11000000081c:	07c4                	addi	s1,sp,964
    11000000081e:	0000                	unimp
    110000000820:	1100                	addi	s0,sp,160
    110000000822:	0000                	unimp
    110000000824:	0420                	addi	s0,sp,520
    110000000826:	0000                	unimp
    110000000828:	1100                	addi	s0,sp,160

Since clint_Mtbl is not on a 64 bit boundary this fails.

InspireSemi, 2023-04-09:
I think the same problem shows up in misaligned-ld/lh/etc and also in the misaligned-sd/sh/etc..

allenjbaum, 2023-04-09:
Yes, that's a bug. The table should be aligned. Need to force table alignment with .align, calculate offset 4 beyond, and realign by clearing low  2 bits of offset

commit 1da0f46a (2023-04-19): Update LI macros, Added infra for VM traps, fixed #313, fixed #332