riscv/riscv-arch-test #334: Zc* arch-test support

ZenithalHourlyRate, 2023-04-14:
Hi, I have opened a [GSoC project for implementing Zc* RTL in rocket chip](https://github.com/chipsalliance/ideas/blob/main/gsoc-2023-ideas.md#implement-zc-risc-v-isa-extension-in-rocket-chip), and there are already candidates for that.

I specifically asked the mentee to use arch-test to test their RTL, as this is the standard way and there are already some arch-test infra in rocket chip, with the expectation that there will be Zc* arch-test when the project starts.

I heard from @liweiwei90 that there are already some WIP for Zc* in arch-test (did not know where the WIP branch is), so I open this issue to track the status.

Can an usable branch be present when the GSoC project starts? Or I need to plan to ask the mentee or someone else to implement these tests.

allenjbaum, 2023-04-14:
Look at item #25:

https://github.com/orgs/riscv-admin/projects/2/views/4

I believe it is blocked because of a Sail issue at the moment

UmerShahidengr, 2024-12-30:
All Zc* tests have been added now, so closing this issue.

closed 2024-12-30 by UmerShahidengr