riscv/riscv-arch-test #336: Warning messages about missing CSRs from new macro code in arch_test.h

ssecatchseagate, 2023-04-26:
There's no explanation about how they should be set. 
Ie, if I create them and set them to empty, what does this do to the test sequence? Is this acceptable functionality?

riscv-test-suite/env/arch_test.h: Assembler messages:
riscv-test-suite/env/arch_test.h:749: Warning: RVMODEL_CLR_MSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
riscv-test-suite/env/arch_test.h:753: Warning: RVMODEL_CLR_MTIMER_INT not defined. Executing this will end test. Define an empty macro to suppress this warning
...

arch_test.h:
//==============================================================================
// Helper macro to set defaults for undefined interrupt set/clear
// macros. This is used to populated the interrupt vector table
//==============================================================================
//****************************************************************
#define RVTEST_DFLT_INT_HNDLR      j cleanup_epilogs
        //Mmode interrupts
#ifndef RVMODEL_SET_MSW_INT    
        .warning "RVMODEL_SET_MSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning"
        #define  RVMODEL_SET_MSW_INT     RVTEST_DFLT_INT_HNDLR   
#endif
...

allenjbaum, 2023-04-26:
These macros are only used during interrupt tests, and are used to clear
the interrupts so they don't keep interrupting after returning from the
trap handler.
If you aren't running interrupt tests, and don't define it: you get a
warning, and the tests running with no issues
If you aren't running interrupt tests, and do define it, the warning is
suppressed, macro is never run, so it pretty much doesn't matter what the
definition is .
If you are running interrupt tests and don't define it - the test will en
prematurely
If you are running interrupt tests and define it as empty- then upon return
from the trap handler it is likely to interrupt again, so no forward
progress.
If you are running interrupt tests and define it as a write to some model
specific address that causes the interrupt wire to be deasserted, the test
continues.

davidharrishmc, 2023-05-14:
This issue threw me and my students because riscof suddenly started producing lots of warnings.

To be very concrete, I've made the warnings go away by adding the following #defines before the #endif in the following two files:

riscof/sail_cSim/env/model_test.h
riscof/spike/env/model_test.h

#define RVMODEL_CLR_MSW_INT
#define RVMODEL_CLR_MTIMER_INT
#define RVMODEL_CLR_MEXT_INT
#define RVMODEL_SET_SSW_INT
#define RVMODEL_CLR_SSW_INT
#define RVMODEL_CLR_STIMER_INT
#define RVMODEL_CLR_SEXT_INT
#define RVMODEL_SET_VSW_INT
#define RVMODEL_CLR_VSW_INT
#define RVMODEL_CLR_VTIMER_INT
#define RVMODEL_CLR_VEXT_INT

allenjbaum, 2023-05-15:
That’s fine as long was you’re not actually running interrupt tests that
might need to use them. Then- depending on the HW implementation - tests
would fail because the model won’t know how to clear the interrupts.
The other alternative is to just remove the warning messages.

PR #352 (2023-07-14): Trap Handler and Test Macros Update
<FOR DOC UPDATES FILL ONLY THE DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

This PR - co-authored by @allenjbaum :
- Resolves issues/bugs and adds requested features in the Trap handler. 
- Adds test macros for testing Virtual Memory in Sv32 mode. [(sample test)](https://github.com/Abdulwadoodd/riscv-arch-test/blob/vm_dev/riscv-test-suite/rv32i_m/vm/address_trans.S)

### Related Issues
- Resolves #338
- Resolves #349 
- Resolves #337 
- Resolves #339 
- Resolves #342 
- Resolves #336 

##
closed 2023-07-14 by pawks