riscv/riscv-arch-test #337: Incorrect macro for register save in arch_test.h

ssecatchseagate, 2023-04-26:
Error: unrecognized opcode `sigupd( x1,x1)'

arch_test.h:
.macro RVTEST_SAVE_GPRS BASEREG REG_SV_ADDR     // optionally save GPRs
.option push
.option norvc
.set offset,0
  LA(     \BASEREG, \REG_SV_ADDR)               //this destroys basereg, but saves rest
  SIGUPD( \BASEREG, x1)
...
  
  Changing these to RVTEST_SIGUPD allowed compile to proceed

allenjbaum, 2023-04-26:
Sigh -  if it isn't tested, it's borken.
Will be fixed in the next merge.
Because registers may hold addresses that will different between DUT and
reference models,
we expect that this will only be used to debug failing code.
We could add an extra parameter to this to mark which registers  do/don't
hold addresses, and skip storing if its an address to make it more useful

ssecatchseagate, 2023-04-26:
> This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi Allen,

I do think this is useful.
While, all these in theory should get echo'd into the signature file already as part of the regular checking, an additional check is useful that other registers didn't corrupted, since it's "free".*
And if I'm going to be debugging a test by putting this in halfway, I don't want false negatives if I'm already debugging stuff, so I wanted it working already for the "easy case"
*Locally, the past version, I had masked out any registers that held test only values to ensure that they matched.

 -Stacey

Seagate Internal

PR #352 (2023-07-14): Trap Handler and Test Macros Update
<FOR DOC UPDATES FILL ONLY THE DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

This PR - co-authored by @allenjbaum :
- Resolves issues/bugs and adds requested features in the Trap handler. 
- Adds test macros for testing Virtual Memory in Sv32 mode. [(sample test)](https://github.com/Abdulwadoodd/riscv-arch-test/blob/vm_dev/riscv-test-suite/rv32i_m/vm/address_trans.S)

### Related Issues
- Resolves #338
- Resolves #349 
- Resolves #337 
- Resolves #339 
- Resolves #342 
- Resolves #336 

##
closed 2023-07-14 by pawks