riscv/riscv-arch-test #338: LI macro throwing shift count out of range warnings for RV64

ssecatchseagate, 2023-04-26:
It appears that warning messages are being generated for shifts that are within the LI macro for the 64bit and beyond case, even they don't show up in the code. There is a warning for the very first instruction with a -1,  though the old and new code is exactly the same:

inst_0:
// rs2 == rd != rs1, rs1==x30, rs2==x31, rd==x31, 
// opcode: orn ; op1:x30; op2:x31; dest:x31; op1val:0xffffffffffffffff;  op2val:0x1
TEST_RR_OP(orn, x31, x30, x31, 0x0000000000000000, 0xffffffffffffffff, 0x1, x1, 0, x2)

riscv-test-suite/rv64i_m/B/src/orn-01.S:47: Warning: shift count out of range (-1 is not between 0 and 63)

Old generated code:
<inst_0>:
     fff00f13          	li	t5,-1
     00100f93          	li	t6,1
     41ff6fb3          	orn	t6,t5,t6
     01f0b023          	sd	t6,0(ra)
     
 New generated code:
 <inst_0>:
     fff00f13          	li	t5,-1
     00100f93          	li	t6,1
     41ff6fb3          	orn	t6,t5,t6
     01f0b023          	sd	t6,0(ra)

allenjbaum, 2023-04-26:
This only occurs when code inside a conditional fails the condition (so
isn't actually instantiated)
This is a harmless, but annoying warning; it is only seen when there is a
real error.
This was solved for the RV32 case, and I have a fix pending for the RV64
case, and will be included in the next merge.

pawks, 2023-04-27:
To be fixed soon.

algrobman, 2023-05-05:
I see the same ...

allenjbaum, 2023-05-05:
Could you send the error message? I suspect that maybe I only fixed the -1
case, but not the ==64 case?

PR #352 (2023-07-14): Trap Handler and Test Macros Update
<FOR DOC UPDATES FILL ONLY THE DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

This PR - co-authored by @allenjbaum :
- Resolves issues/bugs and adds requested features in the Trap handler. 
- Adds test macros for testing Virtual Memory in Sv32 mode. [(sample test)](https://github.com/Abdulwadoodd/riscv-arch-test/blob/vm_dev/riscv-test-suite/rv32i_m/vm/address_trans.S)

### Related Issues
- Resolves #338
- Resolves #349 
- Resolves #337 
- Resolves #339 
- Resolves #342 
- Resolves #336 

##