riscv/riscv-arch-test #340: No coverage for unaligned instructions when compressed extensions are present

ssecatchseagate, 2023-04-30:
The C directory does include some instructions as part of it's flow, however, that is a limited minimal set only required to support the C extension. The specification says:

The C extension is compatible with all other standard instruction extensions. The C extension
allows 16-bit instructions to be freely intermixed with 32-bit instructions, with the latter now able
to start on any 16-bit boundary, i.e., IALIGN=16. With the addition of the C extension, no
instructions can raise instruction-address-misaligned exceptions.

Since we are not covering all of the instructions, this is compatibility coverage that is currently being missed.

allenjbaum, 2023-05-01:
I don't think it is necessary to test that all 32bit instructions can be
executed when not naturally aligned.
Thepart of the HW that needs to be exercised is the instruction fetch
logic, and not the execution logic,
so if any 32B instructions follow a16bit instruction and are executed
properly, I can say that the fetch (and decode) logic can handle the case.
And, indeed, the tests are typically something like LI (constant), followed
by the 16bit op, followed by a 32b signature store,
which will alternate between being aligned and unaligned - and that tests
what we need to test.

pawks, 2023-05-01:
I think many instruction sequences in the current tests will already satisfy these conditions, albeit not intentionally. Also, there are tests which test for branches to 2-byte aligned boundaries. These should also cover some of the cases which you are talking about.

ssecatchseagate, 2023-05-01:
Thanks. I know that we've talked about this in the past, but it seemed important to document it. 

@pawks, @allenjbaum, maybe all we need is coverage for these cases, and possibly a "I/E" test in the C directly that has all the 32b branch types? 

I don't see anything specific about alignment in the _c coverage file. It seems like we should have a mod function in there somewhere looking for a 16 bit aligned address- and then then also callouts for all the normal 32b branches with a 16bit aligned address.  And I don't know how to say generically "fetch an instruction on a not 32b boundary"- I believe our syntax is coverage per opcode.

cjal:
  config: 
      - check ISA:=regex(.*RV32.*I.*C.*) 
  opcode: 
    c.jal: 0
  val_comb:
    'imm_val > 0': 0
    'imm_val < 0': 0
    abstract_comb:
      'walking_ones("imm_val", 11,fltr_func =lambda x: (x>=10 and x<2030) or (x<=-8 and x>-2030),scale_func = lambda x: x*2)': 0
      'walking_zeros("imm_val", 11,fltr_func =lambda x: (x>=10 and x<2030) or (x<=-8 and x>-2030),scale_func = lambda x: x*2)': 0
      'alternate("imm_val",11, fltr_func =lambda x: (x>=10 and x<2030) or (x<=-8 and x>-2030) ,scale_func = lambda x: x*2)': 0

UmerShahidengr, 2023-05-02:
From a design point of view, it is necessary to have Unaligned support for implementing C-extension. Compressed instructions can't be executed without Unaligned support, so passing the compressed instructions test will superimpose the fact that unaligned instructions support was available. I don't think this explicit coverage is necessary.

allenjbaum, 2023-05-02:
I don't quite agree; Unaligned support is specifically for data load store, not instruction fetch and execute.
The two are slightly different (e.g data could be 1..7  byte unaligned, and instructions can only be 2 byte unaligned.
It is allowed to have unaligned data support without the C-extension (and it might not even be an unusual case).
However, the existing test to test that unaligned instruction fetches do occur simply by executing isolated compressed ops followed by uncompressed ops. I though what we were discussing is really the coverage of which uncompressed ops are executed unaligned. I don't think we need extensive coverage of that specifically.

However, one significant missing bit of coverage is whether an unaligned op - compressed or uncompressed - trap correctly. This is likely tested by the compressed load/store ops with odd alignment (though not deliberately), but we don't have any tests that trap (for any reason) on unaligned, uncompressed ops.

pawks, 2023-05-04:
> @pawks, @allenjbaum, maybe all we need is coverage for these cases, and possibly a "I/E" test in the C directly that has all the 32b branch types?
These do exist but not under the C directory, rather the privilege directory(tests with prefix misalign-*). 

> And I don't know how to say generically "fetch an instruction on a not 32b boundary"
We do not have a method to define alignment criteria for PC at the moment. I think it is enough to test for 2b aligned target addresses for branches to satisfy compatibility requirements incase of C extensions. Technically speaking, redirections to a 2b boundary do imply the same constraints.

ssecatchseagate, 2023-05-04:
> This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

I don't think those are intended for "correct 2b aligned behavior" right now.

For example, in the misalign-bge-01.S testcase, the trap handler is defined as true, and the only branch operation includes an odd address, not one aligned to a 16bit boundary.

For the compressed case, 16bit boundaries are not misaligned.. they are simply not on 32b boundaries.

-Stacey

Seagate Internal

ssecatchseagate, 2023-05-04:
That wasn't clear that the constant wasn't the constant. I assumed our macros did fancy immediate manipulation, but you are correct- the assembly is 16bit:
bge   a0,a1,2ee

However, the trap handler is enabled in both cases:

RVTEST_CASE(0,"//check ISA:=regex(.*32.*);check ISA:=regex(.*I.*C.*); def rvtest_mtrap_routine=True;def TEST_CASE_1=True;",misalign-bge)

RVTEST_CASE(1,"//check ISA:=regex(.*32.*);check ISA:=regex(.*I.*Zicsr.*); check ISA:=regex(^[^C]+$); def rvtest_mtrap_routine=True;def TEST_CASE_1=True;",misalign-bge)

The test tests for behaviour when branch targets are 2b aligned irrespective of the presence/absence of the C extension. There are 2 RVTEST_CASE lines in the test, both of which describe mutually exclusive conditions. The trap handler should be defined only if C extension is not present and Zicsr is present. The other test case defines trap handler because while checking the initial tests, it was assumed that the trap handler is necessary for all the tests(as a safeguard). That has since been removed from most of the tests but evidently some still carry that artifact.

Note that bit[0] is assumed to be 0 implicitly by the spec and hence its not included in the encoding. There is no legal way to define an "odd" target address for redirection instructions with immediates as offsets. The only way to try to jump to an "odd" target address is when the offset calculation includes a register(jalr, c.jalr etc) and the spec enforces that bit[0] should be hardwired to 0 after offset addition to obtain the target address. These tests are for testing control redirections to 2b addresses.

The tests are named "misaligned" because for the base ISA it is a misaligned case. The same test can be reused for multiple configurations thereby reducing maintenance overheads.

Seagate Internal

jordancarlin, 2026-03-05:
ACT4 Zca tests check compressed branches and jumps to 16-bit offsets. Closing.