riscv/riscv-arch-test #342: Trap and exit code includes non-RV32E compliant registers

ssecatchseagate, 2023-05-01:
The trap code in arch_test.h contains t3, 4, and 5 which are registers x28-x30, and do not exist for RV32E architectures

RVTEST_CODE_END uses t4 in abort_tests

Also, the init register code uses RVTEST_E, but the gpr save register code uses __RV32E__

allenjbaum, 2023-05-01:
I know about that - I thought it was mentioned in some other issue.
I have a fix, but I don't know the ABI for an RV32E to nail down the
details.

animesh211, 2023-05-20:
Also sw x1,tohost,t5 in model_test.h not compatible for rv32e.

allenjbaum, 2023-05-20:
not sure where you're getting model_test.h, but it is intended to be
implementation specific.
If someone is testing an RV32E, then they should provide a version that
doesn't use t5.

animesh211, 2023-05-24:
Ok, So Is there a version for model_test.h that is rv32e compatible?
I am trying RISCOF framework, and getting model_test.h generated in that.
It comes from here ->
[model_test.h](https://github.com/riscv-software-src/riscof/blob/a25e3155d3a62a042a2c5d976b386e3df5874d70/riscof/Templates/setup/model/env/model_test.h#L17)

allenjbaum, 2023-05-25:
If you have an RV32E implementation, you are responsible for generating the
model_test.h file, just as anyone that has an implementation of an version
of the architecture is.

ssecatchseagate, 2023-05-25:
> This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

But if there is an autogenerated template that is suggested to be adapted for use, shouldn't it be the lowest common denominator and be compliant to rv32e, no Zicsr, etc..?

However, I think this requires a new issue for the riscof repo, rather than merging this into the request for riscv-arch-test, since as Allen points out, model_test.h is not provided by riscv-arch-test

-Stacey

Seagate Internal

animesh211, 2023-05-30:
I was suggesting the same. Also asked for same here in riscof repo. - [issue-85](https://github.com/riscv-software-src/riscof/issues/85).
I think it'll be good if that one line in model_test.h has rv32e compliant register as the tests are present in repo.

closed 2023-07-14 by pawks

ssecatchseagate, 2023-07-14:
Thanks! I only have one concern about the maintenance and readability of this code:

#define T1      x6
...
  LREG    T1, code_bgn_off + 2*sv_area_sz(sp)
  
T1 looks really like the ABI register t1, and I can see where the code has all been changed to reflect this "local temporary", but I'm confused why a full self-documenting rename like "TEMP1" or "LTEMP1" or even something more specific like "ACT1" or "EXCEPTHAND1" wasn't used.  T1 seems to imply t1, and it clearly shouldn't. 

Additionally, it seems dangerous to be combining both ABI and gpr names for registers. You can see in that LREG code that the stack pointer register sp (x2) is used. But the redefinition for "T1" uses x6 (which in this case IS actually t1, but of course, is NOT true for t4, the original problem register). I would think our code would use one or the other, but mixing them could get us into trouble in the future.

PR #352 (2023-07-14): Trap Handler and Test Macros Update
<FOR DOC UPDATES FILL ONLY THE DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

This PR - co-authored by @allenjbaum :
- Resolves issues/bugs and adds requested features in the Trap handler. 
- Adds test macros for testing Virtual Memory in Sv32 mode. [(sample test)](https://github.com/Abdulwadoodd/riscv-arch-test/blob/vm_dev/riscv-test-suite/rv32i_m/vm/address_trans.S)

### Related Issues
- Resolves #338
- Resolves #349 
- Resolves #337 
- Resolves #339 
- Resolves #342 
- Resolves #336 

##