riscv/riscv-arch-test #344: Question about `c.ebreak` test

Abdulwadoodd, 2023-05-03:
I'm trying to run the Compatibility tests for a core that implements `RV32IMZicsr_Zifencei` and the same string is defined under ISA label of `<model>_isa.yaml` file:
```
hart_ids: [0]
hart0:
  ISA: RV32IMZicsr_Zifencei
  physical_addr_sz: 32
  User_Spec_Version: '2.3'
  supported_xlen: [32]
  misa:
   reset-val: 0x40001100
   rv32:
     accessible: true
     mxl:
       implemented: true
       type:
           warl:
              dependency_fields: []
              legal:
                - mxl[1:0] in [0x1]
              wr_illegal:
                - Unchanged
     extensions:
       implemented: true
       type:
           warl:
              dependency_fields: []
              legal:
                - extensions[25:0] bitmask [0x0001100, 0x0000000]
              wr_illegal:
                - Unchanged
```
But when I run the tests using RISCOF command by pointing the `--suite` flag to `rv32i_m`:
```
riscof run --config=config.ini --suite=riscv-arch-test/riscv-test-suite --env=riscv-arch-test/riscv-test-suite/env 
```

RISCOF picks and run the `c.ebreak` test (along with `I`, `M` and `privilege` tests) from the C directory and it shouldn't do that when C is NOT defined as ISA string and misa in isa.yaml. The generated `test_list.yaml` also contains only this test from compressed extension tests with ISA: rv32IC_ZICSR. It has probably something to do with the [conditions](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/C/src/cebreak-01.S#L22) of `c.ebreak` tests which seems correct to me but I'm not sure about its implementation.

Sorry if this isn't a valid question and appreciate any help.

Abdulwadoodd, 2023-05-05:
> The regex should actually be check ISA:=regex(.*I.*C.*Zicsr);

Yes, this fixes the issue. 
Should I raise a PR for this fix?

allenjbaum, 2023-05-05:
I can't see why you shouldn't!

PR #348 (2023-05-08): Fix `c.ebreak` test condition
### Related Issue

This PR resolves #344.

### Description
- Test condition in the `RVTEST_CASE` function is updated. Now, RISCOF picks the `c.ebreak` test only when the `C` (along with `I` and `Zicsr`) extension is enabled.
commit 136ab593 (2023-05-08): Fix `c.ebreak` test condition, resolve #344
closed 2023-05-08 by pawks