riscv/riscv-arch-test #346: FP arithmetic instructions tests

algrobman, 2023-05-05:
where are FP arithmetic  instruction tests? I only see FP conversion instructions in D and F directories .(?)

allenjbaum, 2023-05-05:
Are you looking in just RV64 directories? My understanding is that RV64 uses the tests in the RV32 directory - they're shared.

pawks, 2023-05-05:
They are automatically set by the macros([link](https://github.com/riscv-non-isa/riscv-arch-test/blob/3caa7840b56aa9db871a35c8daf98e561c7f1084/riscv-test-suite/env/arch_test.h#L194-L206))

algrobman, 2023-05-05:
any news about atomics?

allenjbaum, 2023-05-05:
look at the dev-partners spreadsheet here 
https://github.com/orgs/riscv-admin/projects/2/views/19?pane=issue&itemId=23235596
for updates; nothing to do with this issue.

closed 2023-05-05 by allenjbaum