riscv/riscv-arch-test #347: Super long compilation time

algrobman, 2023-05-05:
Hi,

Can you, guys, advice the best latest 64 toolchain  . Some tests take way too long to compile and postprocess. (fmadd.d.b15-01 for ex): (or it may be the tests are too long)

```
-rw-r--r--  1 grobman  927257314 May  5 11:41 fmadd_d_b15-01.cpp.s
-rw-r--r--  1 grobman 105071323 May  5 12:07 fmadd_d_b15-01.dis
-rwxr-xr-x  1 grobman   15515816 May  5 11:42 fmadd_d_b15-01.elf
-rw-r--r--  1 grobman   15608704 May  5 11:42 fmadd_d_b15-01.o

```
You can see that it took 25min just to disassembly elf ...

I use riscv64-unknown-elf-gcc (GCC) 11.1.0

BTW this test simulates more that 1hour, while others just from a few sec to a few min ..

allenjbaum, 2023-05-06:
Yes, FMA takes a very long time, it has a huge number of coverpoints with
crossproducts to get IBM-like coverage.
We plan to break that test up so each subpart is shorter, but I don't know
if that will make it run any faster.

algrobman, 2023-05-06:
there is one more problem - the sizes of the sources of these tests are more than 60MB. Our git setup does not allow such big files to be pushed to the git server. I had real trouble even remove these files from the DB ... **If you want this coverage - these tests should be spitted into 10 tests each ...**
(at the beginning I was thinking that the tools went to the weeds and it was something wrong with the tests sources).

allenjbaum, 2023-05-06:
This is a duplicate of issue #307.

closed 2023-05-06 by allenjbaum