riscv/riscv-arch-test #349: Getting misaligned exception with 64 bit CPU

algrobman, 2023-05-08:
Hi,

I got misaligned load exception inside exception handler for ebreak tests.

it was in this code in arch_test.h

...
/***************************************************/
/**** shared spcl int/excp dispatcher.          ****/
/**** JR to table in t3+t2, indexed by mcause   ****/
/***************************************************/
spcl_\__MODE__\()handler:               // case table branch to special handler code, depending on mcause
        LA(     t3, clrint_\__MODE__\()tbl)
        // load spcl int/excpt handler dispatch table 
        add     t3, t3, t2              // offset into the correct int/excpt dispatch table
        slli    t2, t5, 3               // index into 8b aligned dispatch entry and jump through it
        add     t3, t3, t2
        LREG    t3, 0(t3)  <<<<<<<<<<<<<< **this one got the exception**
spcl_\__MODE__\()dispatch:
        jr      t3                      // this defaults to resto_\__MODE__\()rtn

/**** this is the table of interrupt clearing routine pointers, which could include special handlers ****/
/**** They default to RVMODEL macros above, which are model supplied, then jump to rtn code          ****/
**?????????? <<< missing .align ??????**
clrint_\__MODE__\()tbl:                 //this code should only touch t2..t6
#ifdef rvtest_vtrap_routine  //  M/S/V/U
        .dword  resto_\__MODE__\()rtn           // int cause  0 is reserved, error
        .dword  \__MODE__\()clr_Ssw_int         // int cause  1  Smode SW int
        .dword  \__MODE__\()clr_Vsw_int         // int cause  2  Vmode SW int
        .dword  \__MODE__\()clr_Msw_int         // int cause  3  Mmode SW int

shouldn't the clrint_\___MODE__\()table have 8 bytes alignement enforcement with  .align ?

allenjbaum, 2023-05-08:
Which version are you using?
The one in the repo has an .align 3 on line 1323.

algrobman, 2023-05-08:
yes, I used DB from ~3 months ago, 
And BTW it works for me with LA()  LA works if the target address is allocated not farer than 2GB from that instruction PC ..

allenjbaum, 2023-05-09:
We just fixed the LI that gave the warning (but not merged yet). LA doesn't
do the same thing that LI does, so it is save.

PR #352 (2023-07-14): Trap Handler and Test Macros Update
<FOR DOC UPDATES FILL ONLY THE DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

This PR - co-authored by @allenjbaum :
- Resolves issues/bugs and adds requested features in the Trap handler. 
- Adds test macros for testing Virtual Memory in Sv32 mode. [(sample test)](https://github.com/Abdulwadoodd/riscv-arch-test/blob/vm_dev/riscv-test-suite/rv32i_m/vm/address_trans.S)

### Related Issues
- Resolves #338
- Resolves #349 
- Resolves #337 
- Resolves #339 
- Resolves #342 
- Resolves #336 

##
closed 2023-07-14 by pawks