riscv/riscv-arch-test #354: All tests failing due to undefined macros??

InspireSemi, 2023-05-22:
All of a sudden all our tests are failing due to these macros being undefined??

 [1;31m   ERROR [0m |  [1;31m<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>: Assembler messages:
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:760: Warning: RVMODEL_CLR_MSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:764: Warning: RVMODEL_CLR_MTIMER_INT not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:768: Warning: RVMODEL_CLR_MEXT_INT   not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:773: Warning: RVMODEL_SET_SSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:777: Warning: RVMODEL_CLR_SSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:781: Warning: RVMODEL_CLR_STIMER_INT not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:785: Warning: RVMODEL_CLR_SEXT_INT   not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:790: Warning: RVMODEL_SET_VSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:794: Warning: RVMODEL_CLR_VSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:798: Warning: RVMODEL_CLR_VTIMER_INT not defined. Executing this will end test. Define an empty macro to suppress this warning
<http://192.168.15.15:8080/job/riscvarchtest_c/ws/riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h>:802: Warning: RVMODEL_CLR_VEXT_INT   not defined. Executing this will end test. Define an empty macro to suppress this warning

We cannot make changes that break things like this.  If a company has regression testing (Jenkins) going (like us) then all of a sudden tests blow up and what they thought was passing is now failing. Causing more work for someone to go figure out what has happened??   

As I have said before the more roadblocks we put in place (intentional or not) the fewer people will be willing to invest the time and $$ to use this to "brand" their SOC.    

Going forward any changes should be backwards compatible or if they will break things a major release should be made with announcements that you have to change things..

allenjbaum, 2023-05-22:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h: Assembler
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:760:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:764:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:768:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:773:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:777:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:781:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:785:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:790:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:794:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:798:
> riscvarchtest/riscv-arch-test/riscv-test-suite/env/arch_test.h:802:

It’s a warning, not an error. I’m not sure why it is indicating that a
warning is an error- unless you are testing interrupts.
You can define them all to be empty macros to remove the messages, just as
the warning indicates.

InspireSemi, 2023-05-23:
Not sure what is going on…  Somewhere between the 12th and the 19th everything broke…

Marc Karasek
Principal Software Engineer
M: 678.770.3788

[A close up of a sign    Description automatically generated]
www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.