riscv/riscv-arch-test #356: Conflict Bit[1] definition of misa CSR

max-starfive, 2023-05-31:
In privileged spec 20211203, the bit[1] of misa CSR is described as "Tentatively reserved for Bit-Manipulation extension". But in latest spec, bit[1] is a reserved bit. Because our CPU followed the old 20211203 spec, bit[1] has the value of 1 which indicates B extension is implemented inside. But now this is regarded as a test failure. It is unclear which privileged spec the RVI20 test supports. 
Will RVI20 test support CPUs which followed old spec?

allenjbaum, 2023-05-31:
We follow ratified specs only; the proper way to deal with “tentatively
reserved” is to hedge your bet and either make the bit RW( and gave your
boot code clear it) or have a hidden fuse or control but that sets it.

In this particular case, follow the errata waiver procedure, and I suspect
( but don’t guarantee) the waiver will be approved for this.

max-starfive, 2023-06-01:
Thanks for your reply. 
Since users maybe affected by the changes of spec or test suite (the code under riscv-test-suite directory), could there be a version control to let users to choose the spec version and the correspondent test suite to certificate for?

allenjbaum, 2023-06-03:
The problem you have is that is there is no ratified spec version that
defines MISA[1] to be set, so even if we did that it wouldn't help you or
anyone else.
In general, there are *no* spec versions at all; there are only ratified
specs, and once ratified they are not changed. There may be additional
extensions,
but those extensions have their own names.

max-starfive, 2023-06-09:
Understood. Thanks a lot.

closed 2023-06-09 by max-starfive