riscv/riscv-arch-test #358: Tests supporting multithreading

animesh211, 2023-06-01:
Hi,
I wanted to check if there is a plan to have tests running on multiple threads.. ?

allenjbaum, 2023-06-03:
We don't have particular plans to test running on multiple threads; our
current charter is to test that a hart is arhcitecturally compatible, and
(from an architectural point of view) there is no difference between
multi-theading and multiple processors. All those details are
micro-architectural, and ACTs are agnostic of micro-architecture.
At some point in the future we may add support for multi-harts,
primarily for memory model and atomic op testing - but this is a topic
under active discussion..

UmerShahidengr, 2024-12-30:
The question was explained by @allenjbaum , so closing this one. You may reopen it anytime if you require any help

closed 2024-12-30 by UmerShahidengr