riscv/riscv-arch-test #359: Using new toolchain (12.2) csr opcode errors.

InspireSemi, 2023-06-02:
The _zicsr and _zifecei options for -march are no longer assumed if you specify i as an option to -march=

For test cases that have any csr opcodes in them 9DUT or Sail) it will fail the compile with a message:
Error: unrecognized opcode csrr a0,mhartid', extension zicsr' required

Currently the python scripts for generating makefile for both the DUT and Sail use as passed in parameter {0} 
-march={0} which for testing C extension translates to -march=rv64ic  

Changing the python to a fixed value of : -march=rv64imafdc_zicsr_zifencei instead of the {0} fixes this issue and allows the test cases to compile w/o errors. 

I do not know if this is the long term fix for this as some cross-compilers may not support the full extension set, based on how they were built.

InspireSemi, 2023-06-02:
> Each test file has the RVTEST_ISA function that defines the ISA string which is picked up by the RISCOF and passed to -march flag.
> And each test has an appropriate ISA string depending on the Instruction OPCODES that the test executes. For example for the C extension tests, the RVTEST_ISA is defined as RV64IC<https://url.avanan.click/v2/___https:/github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv64i_m/C/src/cxor-01.S%23L20___.YXAzOmluc3BpcmVzZW1pOmE6bzpmYTBmNDc0MWUxYjA2OGQ2YWFiNDQ3NmM1YWNlYmIwNjo2OjNlZDA6NWJjMzQzMDRmYjFlNjA1MzZjYjM3ZThlOWVjYWExODdmMWU5ZDQxY2U4MzVkYzM1NmQ3MDJkZTlkNjFhNmQzZTpoOlQ> (which translates to -march=rv64ic), except for c.ebreak test. As c.ebreak raises an exception, it executes CSR opcodes. Hence, zicsr is included in the ISA string for this test (link<https://url.avanan.click/v2/___https:/github.com/riscv-non-isa/riscv-ar%20ch-test/blob/main/riscv-test-suite/rv64i_m/C/src/cebreak-01.S___.YXAzOmluc3BpcmVzZW1pOmE6bzpmYTBmNDc0MWUxYjA2OGQ2YWFiNDQ3NmM1YWNlYmIwNjo2OmFhZTU6Njg2ZTljOTE5YTNlMmUyOWE4MmY2MzEyNzhiYWRlNjAxNTMxNmQzODlkYTczYzlmNTA1MDA1ZDAzYThjN2M2MzpoOlQ>).
>
> So, the test carefully includes only those minimum extensions/features to the ISA string that are required by the test. Since the C extension tests do not execute (or are not required to execute) CSR opcodes (except c.ebreak), so ISA string only includes ic and does not include zicsr. If there is a bug in the implementation (or the C extension is disabled in the implementation), it may cause a trap and tries to execute CSR opcodes, then the toolchain will throw the compilation error because zicsr is not included (and not required to be included by that test)

Example from our DUT that uses csr opcodes in test case:

//RVMODEL_BOOT
#define RVMODEL_BOOT \
.section .text.init; \
.extern _fsignature; \
  csrr a0, mhartid              ;\
  li   a1, 0x00                        ;\
  beq  a1, a0, cont             ;\
55: \
  wfi ;\
  j 55b ;\
cont: ;\
  RVMODEL_IO_INIT;

This allows only 1 hart to proceed to runt he tests and puts any other hart into a wfi state.

Marc Karasek
Principal Software Engineer
M: 678.770.3788

[A close up of a sign    Description automatically generated]
www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

allenjbaum, 2023-06-03:
I think the short description is:
 - you have provided a string (or the riscv-config YAML) that says you
support Zicsr,
 - you require Zicsr in your boot code,
 - but if the test doesn't require Zicsr, it tries to compile the test (and
your boot code) without that string
and therefore the compile fails?

MarcKarasek, 2023-06-03:
Yes

UmerShahidengr, 2024-12-30:
I think this issue was resolved, and all existing tests are working on gcc 12.2.
Closing this one, you may reopen it anytime.

closed 2024-12-30 by UmerShahidengr