riscv/riscv-arch-test #389: What should I do first when using this?

drkome, 2023-09-23:
Are there any examples of this? These kinds of spreads seem overly complex to me. I'm sorry for asking such a stupid question. Do I need to install a program like Verilator while using this? I designed a processor, but I didn't understand what to do using this program. Is there a document explaining what I should do step by step? Since I'm new to these topics, I'd appreciate it if you could answer them, assuming I don't know much.

allenjbaum, 2023-09-23:
Did you start looking through the doc/ directory in the git repo?
e.g. Please refer to the RISCOF’s guide on [Running RISCV-ARCH-TESTS](
https://riscof.readthedocs.io/en/stable/arch-tests.html) to run the
architectural test-suite on your DUT.

UmerShahidengr, 2023-11-01:
These tests are run via RISCOF, if you have written RTL of your own processor, then you have to create a plugin for your processor to make it compatible with RISCOF, you can check [here](https://riscof.readthedocs.io/en/stable/plugins.html) to understand how to make a model plugin.
Also you can check [this repo](https://gitlab.com/incoresemi/riscof-plugins/-/tree/master) to look at the model plugins available for different DUTs. It will give you an idea on how to define a plugin for your own RTL.
For another sample, you can see [this model plugin](https://github.com/ee-uet/UETRV-PCore/tree/main/verif/pcore-plugin) which is simple and can be reused for your RTL if it is running on Verilator.

drkome, 2023-11-10:
Than

> These tests are run via RISCOF, if you have written RTL of your own processor, then you have to create a plugin for your processor to make it compatible with RISCOF, you can check [...]

Thank you very much. Iwill do this. I will search to this. so thanks.

closed 2024-12-30 by UmerShahidengr