riscv/riscv-arch-test #396: e7d4b281 & bf5ca309 in the ref's signature

hyperion009, 2023-10-08:
Hi
it's a first time for me to run a riscv-arch-test. I am using the sail as a reference and find that there are e7d4b281 & bf5ca309 at the begin and end of sail's signature files:
[image]
Is this a placeholder or something else?

thanks

allenjbaum, 2023-10-09:
This are "canary" words which are present to catch errors that go beyond
the bounds of the signature region.
If those words are overwritten, then the framework knows that something
went wrong.
(Assuming that Sail is correct, it would see a signature difference, so it
knows the DUT is acting improperly)

allenjbaum, 2023-10-12:
I believe this answers your question, so I'm closing this. If this explanation isn't adequate, reopen it

closed 2023-10-12 by allenjbaum