riscv/riscv-arch-test #397: RVTEST_ISA for privileged spec 1.12

davidharrishmc, 2023-10-09:
I'm trying to write a custom test in the riscof flow for the mconfigptr CSR.   Right now, my RV_TEST_ISA("RV64I_Zicsr") isn't strong enough to support this CSR.  I can't find documentation of what to put in RV_TEST_ISA to force privileged spec 1.12.  Could anyone tell me the incantation, and could I suggest documenting in TestFormatSpec.adoc where to find what ISA strings are supported?

Specifically, when I run readelf on the ref.elf, I get Tag_RISCV_priv_spec_minor: 11, while mconfigptr is part of private spec 1.12.  Then Sail produces an illegal instruction exception on the csrr mconfigptr.

Thank you!

David

UmerShahidengr, 2023-10-12:
I think you are getting this error because Sail doesn't have the support of mconfigptr. For any new CSR, you just need to define it in encoding.h file with its correct address, and give the ISA string in RVTEST_ISA() for compiling/selecting the test for execution. 
The tests are getting compiled and getting selected on Sail with this RVTEST_ISA string, but it is not running on Sail because of the unsupported CSR.

UmerShahidengr, 2023-10-12:
David,
Sail doesn't fully support 1.12 yet. There are a plenty of things to be done. 
For mconfigptr, there is an outstanding PR at Sail repo ([here](https://github.com/riscv/sail-riscv/pull/293)) which is open for quite a while.

closed 2023-10-12 by davidharrishmc