riscv/riscv-arch-test #399: mtrap_sigptr issue

UmerShahidengr, 2023-10-12:
While testing the trap handler in delegated trap (within S-mode), a bug is found in handling mtrap_sigptr.
Trap Handler is managing mtrap_sigptr while saving its physical address in Mmode save area, but when trap handler works in S-mode with virtualization enabled, then physical address of mtrap_sigptr will be translated to virtual address, and that virtual address is needed to be stored in Smode save area.
It will originate a new issue in handling the offset between physical address of mtrap_sigptr and virtual address of mtrap_sigptr. Trap handler has to keep the track of offset value (i.e, the data stored in signature after every trap occurrence), and keep updating the physical and virtual addresses of mtrap_sigptr value in Mmode and Smode (and in Vmode in future) save area respectively.

allenjbaum, 2023-10-13:
What I thought was happening was that Smode (or VS mode) accesses the physical address of mtrap_sigptr (which is the only one that is used) and pre-increments it, It then should relocate it to Smode/VSmode address by adding 
 SSig_Begin-MSigBegin, and uses that to do the actual signature updates. So it is never stored in the Smode save area, it is always translated on the fly.

UmerShahidengr, 2023-11-01:
The issue has been resolved by the PR #401 , so closing it.

closed 2023-11-01 by UmerShahidengr