riscv/riscv-arch-test #400: How to generate tests for RV64D and RV64F using riscv-ctg?

riscvzbs, 2023-10-17:
When I use 'riscv_ctg -v debug -d ./tests/ -r -cf ./sample_cgfs/dataset.cgf -cf ./sample_cgfs/rv64i.cgf -bi rv64i -p2,' I can generate tests for RV64I. However, I couldn't find the provided rv64d.cgf and rv64f.cgf. Now I want to generate tests for D and F extensions. How can I do that? Looking forward to your guidance. Thank you very much.

davidharrishmc, 2023-12-19:
I would like to know this too, and especially whether there is an easy way to modify rv64f.cfg to produce half and quad precision tests.

davidharrishmc, 2023-12-25:
@pawks can you comment on whether these configs can be made available?

pawks, 2023-12-25:
All the cgfs are already present under the sample_cgfs folder in the repo([link](https://github.com/riscv-software-src/riscv-ctg/tree/master/sample_cgfs/sample_cgfs_fext)). The cgfs are placed in the directory corresponding to the minimum ISA requirement for the instruction to be included i.e for a RV64FD configuration instructions from all 4 directories will have to be used. This follows the instruction organization in the ISA manual.

davidharrishmc, 2023-12-26:
@pawks Thank you for the speedy reply and sorry I missed the subdirectory. A few more questions if you have time to support this.

1) It looks like riscv_ctg has changed since you generated fadd_b1-01.S June 17, 2022.  For example, now it wants a --base-isa and seems to want --flen rather than --xlen.  I'm having trouble coming up with a usage string that works with RV32F/fadd.s.cgf.  Can you either point me to a string that works, or recommend an older version of riscv_ctg that is compatible with the usage string recorded in fadd_b1-01.S?

2) RV32F/fadd.s.cgf uses some abstract_comb: 'ibm_b... statements. I'm not finding where these are defined in riscv_ctg.  Will riscv_ctg find these combinations somehow, or is there additional information I need outside the riscv_ctg repo?

3) Related to (2), the abstract_comb ibm_b... takes a parameter of 32.  I presume I'll need to change this to 16 and 128 to generate Zfh and Q tests.  If this doesn't just work out of the box, is there any documentation about finding and modifying this abstract_comb?  I'm comfortable with floating-point and willing to do work if I can find what needs adapting.

allenjbaum, 2023-12-26:
I had to search and find the IBM files also; They're in the isac repo.