riscv/riscv-arch-test #414: Hypervisor Extension enable issue

Allen-StarFive, 2023-11-30:
Our DUT hardware supports H extension, so we want to enable MISA.H = 1.
But Reference-sail_c_simulator doesn't support it at present, which causes some test cases to fail.
How to configure yaml files in this case?

Example in attach files:
[Dubhe-90.zip](https://github.com/riscv-non-isa/riscv-arch-test/files/13507589/Dubhe-90.zip)

Test case fail cause analysis:
[image]
In function <Mxcpt_sig_sv> read misa to t1, Our DUT return misa=0x80000000001411AD, which cause branch unsatisfied.
But SAIL always return misa=0x800000000014112D, which doesn't match H extension.

allenjbaum, 2023-12-02:
The short answer is that Hypervisor has not been implemented in Sail yet, so you can't configure it.
The people who understand Sail and the people that understand the H-extension have a very small overlap.
There are some changes being made to the Sail model that should make implementation simpler... but not simple

allenjbaum, 2023-12-04:
> what's your suggestion for next step? Can we just waive these test
> failures?

The effort to add this into Sail is huge, and there are lots of other
extensions backed up behind it.
I would use Spike as a reference to start with, but I don't know if there
are tool chinese to support H-extension, and I donb't know if
riscv-confiig supports it (that is a much easier problem), and because of
all of this, we don't have tests, There are tests out there (riscv-tests
probably has some and they're designed for spike, and Imperas has both
models and tests), but they're not in arch-test format, and the coverage is
likely pretty poor.
I would start there, or with Imperas tests.

Allen-StarFive, 2023-12-26:
I've already run it with spike and file a report to  riscv-arch-test-reports (https://github.com/riscv-non-isa/riscv-arch-test-reports/pull/6), you can take a look.

allenjbaum, 2023-12-27:
Wow, I just reread my original comments, and its full of weird typos.
OK, I can merge this - but note that this didn't actually run any hypervisor tests!