riscv/riscv-arch-test #418: Compiling Zca tests

davidharrishmc, 2023-12-19:
I'm trying to build the new Zcb  tests from PR364.  I need to modify the ISA string in my spike_rv32gc_isa.yaml.  I can't seem to find an ISA string that is accepted by riscof.  I believe I've fully updated riscof, but maybe there is something I've missed.  For example, each of the following ISA strings produces a message like the one below when I add Zca or Zcb. I've tried many permutations.  Has anyone done this successfully?

I'm facing the same difficulty adding Zicboz for the cmo.zero tests.

David Harris

```
  ISA: RV32IMAFDCZicsr_Zifencei_Zba_Zbb_Zbc_Zbkb_Zbs_Zca
  ISA: RV32IMAFDCZicsr_Zifencei_Zba_Zbb_Zbc_Zbkb_Zbs_Zca_Zcb
   ISA: RV32IMAFDCZicsr_Zifencei_Zba_Zbb_Zbc_Zbkb_Zbs_Zcb
 ISA: RV32IMAFDCZcb

ERROR | Error in /home/harris/cvw/tests/riscof/spike/spike_rv32gc_isa.yaml.
   ERROR |  ISA:
   ERROR |   - Input ISA string does not match regex
   ERROR |   - Input ISA string : RV32IMAFDCZcb does not match accepted canonical ordering
   ERROR |  misa:
   ERROR |   - {'reset-val': ['reset value does not match with extensions enabled']}
m
``

Abdulwadoodd, 2023-12-20:
Hi Prof Harris, 
I think this is not the issue with the ISA string for spike. Using an older version of `riscv-config` may cause this error because riscv-config is used to validate the ISA string inside riscof (and it is installed as part of riscof installation). Support for zce is added in riscv-config in [PR129](https://github.com/riscv-software-src/riscv-config/pull/129). Updating the riscv-config (`pip3 install -U riscv_config`) may resolve this error.
This update and the spike ISA string of `RV32IMAFDCZicsr_Zifencei_Zca_Zcb_Zba_Zbb_Zbc_Zbs` works for me (other permutations of string may also work).

Abdulwadoodd, 2023-12-20:
Oh, and the `build` function of riscof python plugin for spike also needs to be updated with the following (after [this](https://gitlab.com/incoresemi/riscof-plugins/-/blob/master/spike_simple/riscof_spike_simple.py?ref_type=heads#L62) line)
```
      self.isa = 'rv' + self.xlen
      if "I" in ispec["ISA"]:
          self.isa += 'i'
      if "M" in ispec["ISA"]:
          self.isa += 'm'
      if "A" in ispec["ISA"]:
          self.isa += 'a'
      if "F" in ispec["ISA"]:
          self.isa += 'f'
      if "D" in ispec["ISA"]:
          self.isa += 'd'
      if "C" in ispec["ISA"]:
          self.isa += 'c'
      if "Zca" in ispec["ISA"]:
          self.isa += '_zca' 
      if "Zcb" in ispec["ISA"]:
          self.isa += '_zcb'  
      if "Zba" in ispec["ISA"]:
          self.isa += '_zba'
      if "Zbb" in ispec["ISA"]:
          self.isa += '_zbb'
      if "Zbc" in ispec["ISA"]:
          self.isa += '_zbc'    
      if "Zbs" in ispec["ISA"]:
          self.isa += '_zbs'
```

davidharrishmc, 2023-12-20:
Abdul,

Thank you for the pointer.  That got the ISA string reading for me.  Do you know if updating riscv_config is documented anywhere?  Maybe it should be in Section 3.2 of

https://riscof.readthedocs.io/en/stable/installation.html#install-riscof

The next issue I’m seeing is that the RVTEST_CASE is missing 32 or 64, and is thus getting compiled for the wrong architecture.  For example, when compiling for  ISA: RV32IMAFDCZicsr_Zicboz_Zifencei_Zca_Zcb_Zba_Zbb_Zbc_Zbkb_Zbs, I get complaints about selecting a test from the rv64_m/C suite.

    INFO | Running Build for Reference
   ERROR | Error in test: /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/clbu-01.S
Test Selected without the relevant extensions being available on DUT.

The issue in the suite is the RV_TEST_CASE LINE.  It was
RVTEST_CASE(0,"//check ISA:=regex(.*I.*Zca.*Zcb.*);def TEST_CASE_1=True;",clbu)

and I think it should be
RVTEST_CASE(0,"//check ISA:=regex(.*64.*I.*Zca.*Zcb.*);def TEST_CASE_1=True;",clbu)

I’ve encountered this issue on many tests, including some recently posted M extension tests that I fixed, and on rv32e_unratified/B/src/ror-01.S, which was listing I instead of E for some of the ISAs.  Could there be something funny about my configuration that is complaining about these extensions when other people are posting the tests and apparently not having troubles?

Finally, it looks like GCC needs updating for Zcb.  Would you know if this dev branch is the only version that handles Zcb, or if anything is going upstream soon?

https://github.com/linsinan1995/riscv-gcc/tree/riscv-gcc-experiment-zcee

David

davidharrishmc, 2023-12-21:
Also, it looks like the cgf hasn't yet been pushed into riscv-ctg.  It would be great to fix the check ISA there so the same problem doesn't show up if the tests are regenerated.

```
// This file was generated by riscv_ctg (https://github.com/riscv-software-src/riscv-ctg)
// version   : 0.11.1
// timestamp : Tue Jun 20 09:44:07 2023 GMT
// usage     : riscv_ctg \
//                  -- cgf //                  --cgf /home/abd/abd-data/riscv-ctg/sample_cgfs/dataset.cgf \
//                  --cgf /home/abd/abd-data/riscv-ctg/sample_cgfs/rv64i_zcb.cgf \
```

davidharrishmc, 2024-04-09:
GCC 13.2.0 supports Zcb but Sail does not yet accept the PR for Zcb.

https://github.com/riscv/sail-riscv/pull/322

davidharrishmc, 2024-04-22:
Sail now has Zcb support.  Note that --enable-zcb must be provided to Sail to use it.  This implies modifying riscof/sail_cSim/riscof_sail_cSim.py to pass this flag.

https://github.com/riscv/sail-riscv/pull/322

See
https://github.com/openhwgroup/cvw/blob/main/tests/riscof/sail_cSim/riscof_sail_cSim.py
for an implementation of passing the flag.

closed 2024-04-22 by davidharrishmc