riscv/riscv-arch-test #421: Wrong satp setup in arch_test.h

ved-rivos, 2023-12-30:
The following code sets up `satp` in `Bare` mode with a bad `PPN` in RV64. For RV32, it sets up a `SV32` mode page table but with a bad `PPN`.

```
//----------------------------------------------------------------------
init_\__MODE__\()satp:
.ifnc \__MODE__ , M                      // if S or VS mode **FIXME: fixed offset frm trapreg_sv?
        LA(     T4, rvtest_\__MODE__\()root_pg_tbl)     // rplc xsatp w/ identity-mapped pg table 
        csrrw   T4, CSR_XSATP, T4
        SREG    T4, xsatp_sv_off(T1)
.endif
```

allenjbaum, 2023-12-31:
Oops, that address should be shifted by the page size.
Could you file a PR to fix that?

ved-rivos, 2023-12-31:
The intent of the code was not clear to me. For RV32, it can be easily fixed as ((ADDR >> 12) | (SATP32_MODE)). For RV64, what should be the mode? Should it just establish Sv39?

ved-rivos, 2024-01-01:
Please see PR #422

ved-rivos, 2024-01-01:
There are few additional issues:
1. The default identity mapped page table does not have A/D bits set so all accesses lead to page faults.
2. There are no user and supervisor virtual memory ranges and so the page table is usable only for supervisor mode.

allenjbaum, 2024-01-02:
1. If there have been no previous accesses, then both the A and D bits
shouldn't be set. They should generate page faults (or not if Zshwadis or
whatever the acronym is now, is implemented)- that is correct behavior.
Tests that want to avoid that will need to go in and manualyl set those
bits before turning on virtual memory.
2. Hmm, if we set the U bit, then someone would have complained just the
opposite. IT's a default, and a test that want to enable Umode access will
need to manually change that for the pages it cares about.

Thanks for the PR I approved it

closed 2024-01-02 by ved-rivos

allenjbaum, 2024-01-02:
I think the merge failed because the changelong wasn't updated. Wold you
mind....?
From now on we should be filing PRs against a dev branch (which hasn't been
created yet)
and the maintainer will be collecting the changes on a monthly cadence and
adding them to the changelog.

ved-rivos, 2024-01-02:
@allenjbaum updated the changelog.