riscv/riscv-arch-test #425: Should cgfs be checked in instead of generated tests

ved-rivos, 2024-01-13:
Currently the tests generated from the cgf files are checked into the repo. However, the cgf used to generate them is not provided. It may be better from a workflow perspective if the cgf we checked into the directories and the tests generated on the fly as part of build process. That would avoid needing re-update the generated files when the generator or the CGFs change. For instance, to add B extension the ISA string needs to change but this change now requires updating a lot of ctg generated files which could be avoided.

allenjbaum, 2024-01-13:
Awkwardly, the .cgf file is checked in... but into a different repo.
Are you suggesting that .cgf file should be moved into the arch-test repo,
and taken out of the ctg repo, and ctg  (and ISAC?) be modified to take
them from there?

We would still need to update ctg files when when new extension use it to
generate new tests, so we'd be trading
      now updating 1 repo (ctg) with both opcode and CGFs for new tests,
and updating 1 repo (arch-test) with the tests themselves
to  then updating  1 repo (ctg) with just  opcode definitions for new
tests, and updating 1 repo (arch-test) with the tests and CGFs.
If we are writing handwritten tests and not using ctg, then only one repo
would be updated when adding tests ( or just adding coverage).
    I don't see how that avoids re-updating generated files when the
generator or CGFs change though, since regenerating effectively starts from
scratch.
   Unless you'rejust adding a few handwritten tests (and new CGFs) when
adding some extra coverage (e.g. the div-by-0 or denorm-roundup-to-norm
cases)

Anyone (else) have strong feelings about this?

ved-rivos, 2024-01-13:
We could standardize on only updating the CTG. The A/B/C...etc folders in arch test would only hold handwritten files. The CTG generated tests would be generated each time CTG is run based on all matching CGF files in the platform configuration. 

> I don't see how that avoids re-updating generated files when the generator or CGFs change

Of course the .S file will be generated each time riscof is run using CTG. We would not check in the generated .S files into the arch test repo.

This would lead to CGF being checked into only one place. The arch-test would need to be update if and only if a handwritten test is being added or an update being made to the environment files.

allenjbaum, 2024-01-15:
Ah ,I was conflating "checking in" with having it as part of the PR commit,
which would be merged (! into the dev branch only, from now on !).
That happens regardless of whether a test is generated by hand or by CTG.

The CTG (or handwritten) .S files will be part of a PR commit to the
arch-test repo,
but maybe that is different than what you're thinking of for "checked-in"?

I'm fine either way; there should only be one copy of the .cgf in one repo,
though it would be nice to have a link in the other as part of the
commit message

ved-rivos, 2024-01-15:
Right. I was referring to the practice of checking in generated .S files to the repo. If we modified riscof to have an additional step to generate the .S files based on the selected CGFs then checking the .S files generated from those CGFs should not be required.

allenjbaum, 2024-01-15:
Generating tests take a significant amount of times and resources (e.g. FP
can take weeks; Vector will be worse, and Matrix will be scary),
and we want to make sure we don't have different tests each time someone
runs a test. Everybody should be running exactly the same tests,
and CTG uses a hash when it generates test, so every time you run it,
you'll get different tests.
So, we will be checking in the tests.
RISCOF isn't generating tests, it is either hand-done, or CTG does it.

allenjbaum, 2024-01-16:
“Hash” should be “seed”

ved-rivos, 2024-01-16:
> RISCOF isn't generating tests, it is either hand-done, or CTG does it.
Right. RISCOF generating was not implied to say it is modified to do that but modified to add a step into Makefile to invoke CTG to do that.

Each random seed leading to a different signature would however be a showstopper issue. I will table this.

closed 2024-01-16 by ved-rivos

allenjbaum, 2024-01-17:
I think we will be moving ahead with changes that put CFGs in the arch-test
coverage directory, and removing them from the CTG repo
(this will also require changes in various places to scripts that think
they know where the cgf files are).