riscv/riscv-arch-test #437: Example models where asserts are actually defined/used?

stkaplan, 2024-02-21:
I'm having trouble understanding how this test suite is actually used in practice. All the examples I can find (including from spike and sail) define the assert macros (like RVMODEL_IO_ASSERT_GPR_EQ) as empty, meaning the tests will never fail! If that is the case, what are you actually testing?

I also don't understand how the F/D tests are supposed to work. For example, in the fadd tests, it looks like correctval is always being set to 0, meaning asserts wouldn't work even if defined! Where is the expected answer supposed to come from?

allenjbaum, 2024-02-22:
The assertion macro is model specific (whcich means if someone wants to use
it to debug a problem, then they can define it). It's used for debugging a
problem with the model, not for reporting results.

The correctrval is in a comment, and FP comments are known to be incorrect,
and not generated, so you can ignore them,

dpetrisko, 2024-02-22:
Hi @stkaplan,

I'm currently trying to do this for my processor. No idea if this is an efficient or correct approach, but might help you get on the right track: https://github.com/black-parrot-sdk/bp-riscv-arch-test/blob/master/blackparrot/env/model_test.h

Assertions: use the scratch register to form a small stack. Implement a minimal comparison and print out a != b or a == b. 

Correctness: after test completion, copy the entire signature out to a file (I use a testbench hook to print). Offline, compare the example signature from Sail model with the DUT signature.

allenjbaum, 2024-02-23:
To answer the specific questions:

> Yes, I'm trying to debug problems with my model -- isn't that what this
> test suite is for?

No, it isn't. The test suite is intended to test that your model is either
compatible, or it isn't.
The assertion macros can help with that, but the test report will tell you
exactly which signature value is incorrect, and you can trace it back from
there.

> And how can I ignore the correctval for FP tests? What is it actually
> testing if there's no expected result?

Until we get resources to write the tool to fill in the correctval for FP
tests, you can't, I'm afraid.
That too was supposed to back-annotate the macro from the Sail signature,
but it never materialized.

stkaplan, 2024-02-23:
I see. I assumed these were supposed to work like the tests from https://github.com/riscv-software-src/riscv-tests -- exiting with error on failures. I didn't understand the whole test signature part until digging in further after you brought it up. So this is mainly designed to run in the RISCOF framework, comparing results of my model against Sail or Spike?

As a side note, what is the relationship between this repo and https://github.com/riscv-software-src/riscv-tests? Is this going to be replacing that one? Both seem to be actively updated, but the tests in this repo seem more comprehensive, supported more extensions than riscv-tests.

closed 2024-03-01 by stkaplan