riscv/riscv-arch-test #440: ecall check fails in CLIC mode

silabs-kjetil, 2024-03-11:
This issue is only seen when using CLIC mode trap handling

https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/env/arch_test.h#L1160

This code is supposed to check for ecall, and exit early when it is detected. However the way the code is doing this does not work correctly when mcause contains extra bits like in the case when CLIC is used. Typically bits 29:28 will contain 0b11 so mcause will somtimes have the value 0x3000000B when read by the handler and the code will not detect that this is actually an ecall with exception code = 0xB.

This issue has also been discussed in the following PR: https://github.com/riscv-non-isa/riscv-arch-test/pull/436#issuecomment-1985661643

allenjbaum, 2024-03-20:
> This issue has also been discussed in the following PR: #436 (comment)

Ah; are there any other bits that might be set for an exception?

allenjbaum, 2024-03-20:
It looks like this check should clear bits XLEN-2..12 before checking .
We may have to make a larger change and store all of mcause instead of just
the upper and lower bits then.

allenjbaum, 2024-03-20:
Actually, no - we already save all of mcause

silabs-kjetil, 2024-03-20:
Yes there might be other bits that are set, so in the case when CLIC is used we have to only look at bits 11:0 to inspect the exception/interrupt code. Here is a picture of the changes done to xcause for CLIC mode.

[image]

It looks like the arch_test.h file already contains a mask to extract the exception cause bits, however this is not used when checking for an ecall.

exception mask: https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/env/arch_test.h#L157

allenjbaum, 2024-03-22:
Ecall is known and is required, so we don't need that mask in this case.
We only do the ecall check when we know it's an interrupt (sign=0).
In any case, the fix is in, and I just need to get someone to test it. We
will check if sign=0 and bits 11:0= 0x8..0xB