riscv/riscv-arch-test #444: jalr-01.S fails with illegal operands on latest binutils 2.42

piotro888, 2024-03-14:
`jalr-01.S` fails to assemble with latest binutils (>=2.42) with the following error:

` riscv-arch-test/riscv-test-suite/rv32i_m/I/src/jalr-01.S:72: Error: illegal operands 'la x0,5b'`

It happens due to `TEST_JALR_OP` macro expanding to load address instruction into `x0` rd in 
```
inst_7:
// rs1==x31, rd==x0, imm_val == 32, 
// opcode: jalr; op1:x31; dest:x0; immval:0x20; align:0 
TEST_JALR_OP(x15, x0, x31, 0x20, x10, 28,0)
```

I found that braking change was introduced in commit https://github.com/bminor/binutils-gdb/commit/27b33966b18ed8bf1701a60999448224b1d28273 on binutils including rejecting `x0` as a valid operand for `la`.

Case with rd==x0 is present in RV32E, RV32I and RV64I

I'm not sure if this problem should be reported to binutils or maybe the test case changed here (the `LA(rd,5b)` inside the macro doesn't seem to be too useful, maybe could be replaced under `.if rd==x0` or always)

allenjbaum, 2024-03-14:
> I found that braking change was introduced in commit bminor/binutils-gdb@
> 27b3396

The test is responsible for generating this inside an LA macro wrapper, so
in the next version of arch_test it will just skip generating code if the
REG argument is X0.
This particular case is should be a perfectly legitimate way of expressing
a JR with JALR, and in fact JALR is converted by the assembler into a JR,
so I'd file this as a issue with binutils.
This test would ensure that the link is not store into any gpr.

piotro888, 2024-06-29:
Test fixed with #446 to workaround binutils issue.

closed 2024-06-29 by piotro888