riscv/riscv-arch-test #445: misalign-jal-01.S without support for compressed instruction signature mismatch

ArbnorSh, 2024-03-17:
The reference model used is Sail, and only the second line of the signature output is different from the DUT.
How to find what is causing this mismatch? All other tests in `privilege` directory are passing (same signature with ref).

```
$ diff 
< 56ff747d
---
> d6ff74dd
```
[ref.signature.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/14627556/ref.signature.txt)
[DUT.signature.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/14627557/DUT.signature.txt)
[misalign-jal-01.log.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/14627562/misalign-jal-01.log.txt)

allenjbaum, 2024-03-21:
I've been trying to see what is going on. I don't know if the log is the DUT or SAIL, but what is being store in the signature is the delta between the address of the beginning of the test, and a pseudo-random value in a register. The problem seems to be that Sail and the DUT have different values as starting addresses. 
This sounds like a test bug, frankly, but I can't be sure unless I can see the other log. I'm a little puzzled that this hasn't turned up earlier, and that is what really odd.

ArbnorSh, 2024-03-21:
@allenjbaum 
The log is SAIL, the DUT does not have such mechanism implemented as SAIL that dumps all the verbose info to a log file. 
The issue seems to be this instruction (from ref) address 0x800004E4 - `sub a0,a0,sp`, I can confirm that `a0` here has the same value in DUT and in SAIL, that value is `0x56FF76DD`, but `sp` differs. And yes you're right, the DUT has instructions located in ROM that start from address 0 and data in RAM that start from address `0x80000000`. 
I am attaching the disassemble for DUT and REF.
[ref_disass.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/14711873/ref_disass.txt)
[dut_disass.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/14711874/dut_disass.txt)

allenjbaum, 2024-03-22:
It's pretty clear that the test is buggy, and we will have to fix it. What
is more pressing to me is why we haven't caught this previously.
I believe we have tested this with different load points, and it had
passed, and it shouldn't have.

ArbnorSh, 2024-03-23:
All implementations I am seeing in github, run these tests with Compressed instruction support. Note in this case DUT does not implement the `C` extension and Sail is being run with `-C or --disable-compressed`. If you have any DUT reference, please share that and I can check this test there!

allenjbaum, 2024-03-26:
Ah, that would explain it. I feel better now.

cr1901, 2025-01-16:
I decided to add the privileged RISCOF tests to [my core](https://github.com/cr1901/sentinel) recently. It doesn't support compressed at all, so I'm running SAIL with `--disable-compressed`. I can duplicate this behavior up to the second line of my DUT and ref signatures differing _only_ by the top level bit. Although the actual signature values are different from @ArbnorSh's, all other lines of both DUT and ref signatures match.

I have uploaded my updated plugin and tests to the [`riscof-priv` branch](https://github.com/cr1901/sentinel/tree/riscof-priv). Assuming you have `pdm` installed (`pipx install pdm`), the following commands _on a Linux system with `GLIBC_2.33` support_ (Ubuntu 22.04 is fine) should regenerate the below files:

```sh
git clone git@github.com:cr1901/sentinel.git
git fetch origin riscof-priv
git checkout riscof-priv
pdm install -G dev -G riscof
pdm riscof
```

I just ran the above commands under `/tmp`, just to check; `diff -u` claims only differences in timestamps and paths for the attached files. So I imagine any other files one would need to look at (`report.html`) are just fine too :). Additionally, my RISCV GCC toolchain is old, so I strip `Zicsr` from the arch before building the test files. Set `strip_zicsr` to `0` in `config.ini` if this breaks things.

[misalign-jal-01.log](https://github.com/user-attachments/files/18440598/misalign-jal-01.log)
[Reference-sail_c_simulator.signature.txt](https://github.com/user-attachments/files/18440480/Reference-sail_c_simulator.signature.txt)
[DUT-sentinel.signature.txt](https://github.com/user-attachments/files/18440605/DUT-sentinel.signature.txt)
[DUT VCD](https://github.com/user-attachments/files/18440621/misalign-jal-01-vcd.zip)

In the interim, is there any way to _not_ run this single test _aside from_ "commenting it out in the output testlist"? I think I'd like to add the privileged tests to CI anyway and add `misalign-jal-01.S` as an expected fail.

cr1901, 2025-01-17:
ref: [ref.disass.txt](https://github.com/user-attachments/files/18453646/ref.disass.txt)
DUT: [misalign-jal-01.disass.txt](https://github.com/user-attachments/files/18453658/misalign-jal-01.disass.txt)

The relevant store happens at the instruction with the opcode "0x00A0A023". On the DUT, the value "0xfffffda2" is stored to the _32-bit-aligned_ address `0x845 << 2`, or `0x2114`. On the ref, the value "0x7FFFFDA2" is stored to the address `0b0010000000000000000011000100010100` or `0x80003114`

allenjbaum, 2025-01-17:
The disassembly doesn't match. This can happen whenever the code is placed
at a different PA between DUT and reference, and in this case the DUT
starts at 0, and the reference starts at 80000000.
Which turns out to be the difference in the signature. The test is supposed
to compensate for that, but the macro is so convoluted, it's hard to figure
out what is going on.

I haven't figured out where exactly the test is going wrong; any
possibility you could send the reference log file?

cr1901, 2025-01-17:
Is the reference log file different from the [misalign-jal-01.log file](https://github.com/user-attachments/files/18440598/misalign-jal-01.log) I provided in my [initial comment](https://github.com/riscv-non-isa/riscv-arch-test/issues/445#issuecomment-2595836256)? If so, what file should I be looking for?

allenjbaum, 2025-01-19:
Hmm: yea it appears it is.. IF that is the reference log file,  do you have
a log file produced from the DUT?

cr1901, 2025-01-19:
> do you have a log file produced from the DUT?

No, I don't have a log file in the form of the one produced by SAIL; the VCD file should serve the same purpose, but optimized for waveform viewer applications.

allenjbaum, 2025-01-19:
Unfortunately, no.  I'm trying to find where the DUT and reference diverge.
Looking at a VCD is pretty useless (for me, anyway).
At some point, a value is calculated in the ra register  (I think), and
that is the one and only value that gets store into the signature. . That's
what I'm trying to find.
Until I can see the point of divergence, I can't tell if the DUT is doing
the wrong thing, or the test is not set up to handle the configuration of
the DUT.
I am fairly sure that is a bug in the test, and that it is using an
uninitialized register: which is fine as long as both DUT and reference are
loaded at the same position.

allenjbaum, 2025-01-19:
Actually, I'm pretty sure I see the bug. The test is storing the delta of 2
addresses, one of which is derived from the link register after a JAL, and
one of which some label.
But, if the JAL traps, then nothing is stored in the link register- and so
instead of a fixed delta, the result is something that depends on the load
addresses.
It's a bit amazing that no one has reported this one before...

The fix is to initialize the link register with a code address.

allenjbaum, 2025-01-19:
specifically, insert " mv rd, tempreg" between lines 617 and 618 (the first
two lines of TEST_JAL_OP)  in tests_macros.h.
Try that, and file a PR if it works.

cr1901, 2025-01-19:
That works/fixes my issue. I filed a PR: #599.

allenjbaum, 2025-01-20:
Thanks so much. This was a tricky one to diagnose.