riscv/riscv-arch-test #448: Zfh's 'flh-align-01' test is executed even if Zfh is not enabled

stnolting, 2024-03-27:
I am using the latest version of this repository for a custom RISCOF port. The test fails because the `Zfh/src/flh-align-01.S` test is executed although the `Zfh` ISA extension is not enabled.

* My test run of RISCOF: https://github.com/stnolting/neorv32-riscof/actions/runs/8446548698/job/23135504941#step:8:234
* The port's ISA config: https://github.com/stnolting/neorv32-riscof/blob/main/plugin-neorv32/neorv32_isa.yaml#L3

I am not sure, but maybe there is a typo in the `lh-align` test's regex list:

https://github.com/riscv-non-isa/riscv-arch-test/blob/8a0cdceca9f0b91b81905eb8497f6586bf8d1c6b/riscv-test-suite/rv32i_m/Zfh/src/flh-align-01.S#L32

which (I think) should also include `F` and `Zfh` as in the remaining `Zfh` test cases:

https://github.com/riscv-non-isa/riscv-arch-test/blob/8a0cdceca9f0b91b81905eb8497f6586bf8d1c6b/riscv-test-suite/rv32i_m/Zfh/src/flt_b1-01.S#L32

davidharrishmc, 2024-04-08:
flh_align-01.S contains 

RVTEST_ISA("RV32IF_Zicsr_Zfh,RV64IF_Zicsr_Zfh")

When I build the tests with riscof, this prevents running this test on a system without Zfh.

If you are building the tests with a different process that examines RVTEST_CASE rather than RVTEST_ISA, maybe the RVTEST_CASE should be more complete.  It should be harmless to make RVTEST_CASE more comprehensive.

I'd suggest you make a PR with the change.  If it's hard for you to make this PR, I can make it for you.

davidharrishmc, 2024-04-08:
The mistaken assertion is that not all tests have the same RISCV_ISA and RVTEST_CASE parameters. flh is an outlier, which is causing the problem.

flh-align-01 has

```
RVTEST_ISA("RV32IF_Zicsr_Zfh,RV64IF_Zicsr_Zfh")
...
RVTEST_CASE(0,"//check ISA:=regex(.*I.*);def TEST_CASE_1=True;",flh-align)
```

while other test cases such as flt_b1-01.S has

```
RVTEST_ISA("RV32IF_Zicsr_Zfh,RV64IF_Zicsr_Zfh")
...
RVTEST_CASE(0,"//check ISA:=regex(.*I.*F.*Zfh.*);def TEST_CASE_1=True;",flt_b1)
```

allenjbaum, 2024-04-08:
Huh; I thought I looked at the other tests and they were the same.
And I misunderstood David's comment - I thought he was saying something else was going on.
File the PR for this change (one parameter in the RVTEST_CASE macro) , and we will try to get that done quickly.
I think we need that done on the dev branch (and we will probably need to ensure that all other PRs outstanding on the main Branch get moved to the dev branch.

stnolting, 2024-04-09:
I'll do a PR.
Zfh is not yet available on the dev branch so I'll make a PR targeting the main branch.

allenjbaum, 2024-04-09:
We clearly need a cleanup on the dev branch & start doing release from the
dev branch, so no more PRs to the main branch.
There's too much stuff in flight right now.

closed 2024-04-16 by stnolting