riscv/riscv-arch-test #450: RVTEST_CODE_END use of RVTEST_SAVE_GPRS in env/arch_test.c doesn't work

ssecatch-w, 2024-04-03:
Assembler messages:
 Error: unrecognized opcode `rvtest_save_gprs x1 gpr_save'
 
 The line with the error is 
 RVTEST_CODE_END
 
 looking at the definition:
 .macro RVTEST_CODE_END          // test is ended, but in no particular mode
  .option push
  .option norvc
  .global rvtest_code_end       // define the label and make it available
  .global cleanup_epilogs       // ****ALERT: tests must populate x1 with a point to the end of regular sig area
/**** MPRV must be clear here !!! ****/
rvtest_code_end:                // RVMODEL_HALT should get here
  #ifdef rvtest_gpr_save        // gpr_save area is instantiated at end of signature
    RVTEST_SAVE_GPRS  x1        gpr_save
  #endif
    RVTEST_GOTO_MMODE           // if only Mmode used by tests, this has no effect
cleanup_epilogs:                // jump here to quit, will restore state for each mode

RVTEST_SAVE_GPRS used to be a macro:
.macro RVTEST_SAVE_GPRS BASEREG REG_SV_ADDR     // optionally save GPRs

But now it's been replaced with a define which doesn't match the call
#define RVTEST_SAVE_GPRSM(_BR, _LBL, ...)               ;\

allenjbaum, 2024-04-03:
Looks like an extra character got introduced in the macro name.
Try deleting the "M"  in  "#define RVTEST_SAVE_GPRSM"
(line 633 I think) and see if that fixes it.

ssecatch-w, 2024-04-03:
Same error. I suspect the problem is that that macro definition expects at least two parameters (if not more). 

Out of curiosity, I fixed the "call":
 RVTEST_SAVE_GPRS(x1, gpr_save)
 
 And that does compile, though I haven't tested to see if it works yet functionally

allenjbaum, 2024-04-03:
That's what I thought, but ... where is arch_test.c??
There are no .c files in the arch_test repo that come to mind

ssecatch-w, 2024-04-03:
My apologies! arch_test.h 
https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/env/arch_test.h